ros2/rmw_zenoh #490: Realsense camera node OpenGL problem with rmw_zenoh

Yeisonint, 2025-02-28:
Using the CycloneDDS implementation, the realsense camera node works as expected

`RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`

```bash
# ros2 run realsense2_camera realsense2_camera_node
[INFO] [1740759470.807820571] [camera.camera]: RealSense ROS v4.56.3
[INFO] [1740759470.808084255] [camera.camera]: Built with LibRealSense v2.56.3
[INFO] [1740759470.808223553] [camera.camera]: Running with LibRealSense v2.56.3
[INFO] [1740759471.623790718] [camera.camera]: Device with serial number ########## was found.

[INFO] [1740759471.623973376] [camera.camera]: Device with physical ID 2-1.2.4-30 was found.
[INFO] [1740759471.624009889] [camera.camera]: Device with name Intel RealSense D435I was found.
[INFO] [1740759471.629369551] [camera.camera]: Device with port number 2-1.2.4 was found.
[INFO] [1740759471.630428638] [camera.camera]: Device USB type: 3.2
[INFO] [1740759472.285295614] [camera.camera]: getParameters...
[INFO] [1740759472.334555273] [camera.camera]: JSON file is not provided
[INFO] [1740759472.335660409] [camera.camera]: Device Name: Intel RealSense D435I
[INFO] [1740759472.336433957] [camera.camera]: Device Serial No:  ##########
[INFO] [1740759472.337196208] [camera.camera]: Device physical port: 2-1.2.4-30
[INFO] [1740759472.337962939] [camera.camera]: Device FW version: 5.16.0.1
[INFO] [1740759472.338651525] [camera.camera]: Device Product ID: 0x0B3A
[INFO] [1740759472.339344847] [camera.camera]: Sync Mode: Off
[INFO] [1740759479.955472010] [camera.camera]: Stopping Sensor: Depth Module
[INFO] [1740759479.955738574] [camera.camera]: Stopping Sensor: RGB Camera
[INFO] [1740759479.955832367] [camera.camera]: Stopping Sensor: Motion Module
[INFO] [1740759480.303357986] [camera.camera]: Starting Sensor: Depth Module
 28/02 16:18:00,447 WARNING [281472183277872] (messenger-libusb.cpp:42) control_transfer returned error, index: 1, error: Resource temporarily unavailable, number: 11
 28/02 16:18:00,448 WARNING [281472183277872] (messenger-libusb.cpp:42) control_transfer returned error, index: 1, error: Resource temporarily unavailable, number: 11
...

```

But Using the Zenoh RMW the node fails showing a problem with OpenGL

`RMW_IMPLEMENTATION=rmw_zenoh_cpp`

```bash
# ros2 run realsense2_camera realsense2_camera_node
[INFO] [1740759653.689146562] [camera.camera]: RealSense ROS v4.56.3
[INFO] [1740759653.689356486] [camera.camera]: Built with LibRealSense v2.56.3
[INFO] [1740759653.689410631] [camera.camera]: Running with LibRealSense v2.56.3
[INFO] [1740759654.521702996] [camera.camera]: Device with serial number ########## was found.

[INFO] [1740759654.521918840] [camera.camera]: Device with physical ID 2-1.2.4-30 was found.
[INFO] [1740759654.521991386] [camera.camera]: Device with name Intel RealSense D435I was found.
[INFO] [1740759654.526420720] [camera.camera]: Device with port number 2-1.2.4 was found.
[INFO] [1740759654.533640222] [camera.camera]: Device USB type: 3.2
[ERROR] [1740759655.158057440] [camera.camera]: Error starting device: Could not open OpenGL window, please check your graphic drivers or use the textual SDK tools
^C[ros2run]: Received signal:  Interrupt
[INFO] [1740759661.391788520] [rclcpp]: signal_handler(signum=2)
[INFO] [1740759661.392595416] [rclcpp]: signal_handler(signum=2)

```
Why would changing the implementation show a problem of this type?

Yadunund, 2025-03-07:
> Why would changing the implementation show a problem of this type?

It shouldn't and the error above does not indicate that the issue has anything to do with the RMW. Still, it would be good to follow [these instructions](https://github.com/ros2/rmw_zenoh?tab=readme-ov-file#terminate-ros-2-daemon-started-with-another-rmw) when switching RMWs.

Anyways, I have not been able to reproduce this issue with latest jazzy bianries on Ubuntu 24.04. 

Could you provide more details of your setup? ROS verison, how you installed ROS, OS version, how you installed rmw_zenoh, etc.

Yadunund, 2025-03-19:
Closing this given lack of information. But feel free to reopen.