ros2/rmw_zenoh #493: Pass All Tests in ros2.repos

YuanYuYuan, 2025-03-03:


JEnoch, 2025-03-03:
Note: _"TLS Issue"_ refers to the issue with **Thread Local Storage** discussed in https://github.com/ros2/rmw_zenoh/issues/170. 
The fix is to ensure to call `rclcpp::shutdown()` when a test exits, as recommended in [README.md](https://github.com/ros2/rmw_zenoh/blob/eccfe4d048f587d136482616c1d1974df3b5625c/README.md#crash-when-program-terminates)

Yadunund, 2025-03-03:
@YuanYuYuan please run tests using this ros2.repos file: https://github.com/ros2/ros2/pull/1649

YuanYuYuan, 2025-03-04:
> @YuanYuYuan please run tests using this ros2.repos file: https://github.com/ros2/ros2/pull/1649 

Sure. 

> Also you can leave fixing  test_rmw_implementation to me!

Thanks 🙂

Yadunund, 2025-03-06:
@YuanYuYuan could we update the table above to strikeout the tests that have been fixed?

Yadunund, 2025-03-07:
test_rmw_implementation has been resolved #481 

Remaining test failures:

| name                    | list                                                               | reason                                                       |
| ----------------------- | ------------------------------------------------------------------ | ------------------------------------------------------------ |
| rosbag2_transport       | 1 - test_play__rmw_zenoh_cpp (Timeout)                             | #510                                                         |

YuanYuYuan, 2025-03-13:
Some pytest failures were not caught last time. Here's the update.

| package         | reason                                       |
| --------------- | -----                                        |
| ros2action      | issue: RUST_LOG                              |
| ros2lifecycle   | issue: RUST_LOG                              |
| ros2param       | issue: RUST_LOG                              |
| ros2doctor      | https://github.com/ros2/ros2cli/pull/985     |
| ros2node        | https://github.com/ros2/rmw_zenoh/pull/536   |
| ros2service     | https://github.com/ros2/rmw_zenoh/pull/536   |
| ros2topic       | https://github.com/ros2/rmw_zenoh/issues/523 |
| sros2           | issue:  ros2 security generate_policy: error |

Yadunund, 2025-03-19:
@YuanYuYuan and @ahcorde what the the remaining test failures you are observing?

YuanYuYuan, 2025-03-19:
> [@YuanYuYuan](https://github.com/YuanYuYuan) and [@ahcorde](https://github.com/ahcorde) what the the remaining test failures you are observing?

On Ubuntu rolling, except for the sros2, everything is fine.
I'm testing on Windows rolling.

Yadunund, 2025-03-19:
@YuanYuYuan could we open a ticket with more details on the sros2 failures?

YuanYuYuan, 2025-03-19:
> [@YuanYuYuan](https://github.com/YuanYuYuan) could we open a ticket with more details on the sros2 failures?

No worries. The error log said "connection refused", which is different from the one I saw last time. The test can pass if we run it individually.

YuanYuYuan, 2025-03-31:
Superseded by #567.

closed 2025-03-31 by YuanYuYuan