ros2/rmw_zenoh #510: [Test Failure] rosbag2_transport / test_play__rmw_zenoh_cpp / RosBag2PlayQosOverrideTestFixture.topic_qos_profiles_overridden_incompatible

YuanYuYuan, 2025-03-06:
## Error log

```console
4: [ RUN      ] RosBag2PlayQosOverrideTestFixture.topic_qos_profiles_overridden_incompatible
4: 2025-03-06T15:49:03.678243Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: b9a9de20a9d914a52e92b39059be77a2
4: 2025-03-06T15:49:03.679396Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[::1]:40019
4: [INFO] [1741276143.705926716] [rosbag2_player]: Set rate to 1
4: [INFO] [1741276143.706111232] [rosbag2_player]: Overriding QoS profile for topic /test_topic
4: [INFO] [1741276143.716725815] [rosbag2_player]: Adding keyboard callbacks.
4: [INFO] [1741276143.717198713] [rosbag2_player]: Press SPACE for Pause/Resume
4: [INFO] [1741276143.717317767] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
4: [INFO] [1741276143.717381757] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
4: [INFO] [1741276143.717420830] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
4: [INFO] [1741276143.717482065] [rosbag2_player]: Playback until timestamp: -1
4: /workspace/ws-whole-rolling/src/ros2/rosbag2/rosbag2_transport/test/rosbag2_transport/test_play.cpp:1183: Failure
4: Expected equality of these values:
4:   result
4:     Which is: 4-byte object <00-00 00-00>
4:   std::future_status::timeout
4:     Which is: 4-byte object <01-00 00-00>
4:
4: Error. Can't return old SIGINT handler, someone override our signal handler
```

Mallets, 2025-03-13:
Running the test on my machine I had a blocking timeout issue.

```
[ RUN      ] RosBag2PlayTestFixture.recorded_messages_are_played_for_topics_and_services
2025-03-13T10:26:54.778877Z  WARN ThreadId(06) zenoh::net::runtime::orchestrator: Scouting delay elapsed before start conditions are met.
[WARN] [1741861614.779171919] [rmw_zenoh_cpp]: Unable to connect to a Zenoh router. Have you started a router with `ros2 run rmw_zenoh_cpp rmw_zenohd`?
[WARN] [1741861614.779551712] [rmw_zenoh_cpp]: Unable to connect to a Zenoh router after 1 attempt(s). Please ensure that a Zenoh router is running and can be reached. You may increase the number of attempts to check for a router by setting the ZENOH_ROUTER_CHECK_ATTEMPTS environment variable. Proceeding with initialization but other peers will not discover or receive data from peers in this session until a router is started.
[INFO] [1741861614.800186927] [rosbag2_player]: Set rate to 1
[INFO] [1741861614.808572239] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1741861614.808681447] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1741861614.808697281] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1741861614.808710072] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1741861614.808723281] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
[INFO] [1741861614.809472283] [rosbag2_player]: Playback until timestamp: -1
[WARN] [1741861624.800049080] [subscription_manager_1741861614788641608]: SubscriptionManager::continue_spinning(..) finished by timeout

0% tests passed, 1 tests failed out of 1
```

This issue seems to be solved by https://github.com/ros2/rmw_zenoh/pull/537.

All the test suite now runs to completion without getting stuck.
However, I still get the following failure (same as the issue described above):
```
[ RUN      ] RosBag2PlayQosOverrideTestFixture.topic_qos_profiles_overridden_incompatible
[INFO] [1741861862.500200618] [rosbag2_player]: Set rate to 1
[INFO] [1741861862.500388368] [rosbag2_player]: Overriding QoS profile for topic /test_topic
[INFO] [1741861862.504104330] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1741861862.504808914] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1741861862.504847789] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1741861862.504874081] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1741861862.504886748] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
[INFO] [1741861862.504917664] [rosbag2_player]: Playback until timestamp: -1
/home/luca/rosbag2/rosbag2_transport/test/rosbag2_transport/test_play.cpp:1183: Failure
Expected equality of these values:
  result
    Which is: 4-byte object <00-00 00-00>
  std::future_status::timeout
    Which is: 4-byte object <01-00 00-00>
```

YuanYuYuan, 2025-03-13:
The 2nd one will be resolved by https://github.com/ros2/rosbag2/pull/1936.

YuanYuYuan, 2025-03-19:
Let's close the issue since https://github.com/ros2/rosbag2/pull/1936 has been merged.

closed 2025-03-19 by YuanYuYuan