ros2/rmw_zenoh #522: Migration from `zenoh-plugin-ros2dds` to `rmw_zenoh`

meetgandhi-dev, 2025-03-10:
Hello,

First, thank you for the excellent work!

We have begun exploring the Zenoh integration with ROS2 using the [zenoh-plugin-ros2dds](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds) and are now planning to switch to `rmw_zenoh` with the `ROS 2 Jazzy` distribution.

In our use case, we only want a specific group of topics and services to be accessible outside our robot, using the ROS2 DDS configuration provided [here](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/blob/main/DEFAULT_CONFIG.json5#L11). I’ve gone through the documentation and configuration in this repository but couldn’t find any relevant configuration for this.

Am I missing something here, or is this feature still to be implemented?

JEnoch, 2025-03-10:
Hi @meetgandhi-dev ,

You didn't miss anything, it's not yet documented (I plead guilty to not have made progress on [this](#395) so far...).

For what you want, you can configure Zenoh ACL, as explained here:
https://github.com/ZettaScaleLabs/roscon2024_workshop/blob/main/exercises/ex-6.md

meetgandhi-dev, 2025-03-12:
Hi @JEnoch ,
Thank you for the prompt response and providing the example file.
It looks like we can allow a list of specific keys on a dedicated interface. Just wanted to confirm if there's a way to add a ROS namespace for topics that are going outside the system? something similar to [this](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/blob/main/DEFAULT_CONFIG.json5#L26)?

JEnoch, 2025-03-13:
No, at this stage there is no way for the Router to add a ROS namespace on the fly.
The Router is unaware of how `rmw_zenoh` structures its key expressions, so it cannot insert a ROS namespace at the expected position within.

aaronchongth, 2025-03-25:
Thanks for setting up some information here. We are also looking into migrating from `zenoh-bridge-ros2dds` to `rmw_zenoh`, where the main feature of the bridge that we have been using was to connect multiple systems (running on different `ROS_DOMAIN_ID`s so they don't interfere with each other), to a central system orchestrator, like so, https://github.com/osrf/nexus/blob/main/docs/media/nexus_architecture.png.

Just in case I missed any updates or was unaware of features, I was curious what would be the best approach, especially with these constraints,
* each workcell should ideally not be reachable by other workcells, since they have common topics like `/tf` and `/tf_static` and will interfere with operations. (hence the separation of `ROS_DOMAIN_ID` and using bridges previously)
* but at the same time, each workcell does have common services and actions that the central system orchestrator needs to work with `/workcell_request` for example, which kinda makes separating by domain ID not ideal, since the zenoh key expressions will not match

Let me know if missed anything obvious! Any ideas would be greatly appreciated.