ros2/rmw_zenoh #523: [Test Failure] ros2 cli

ahcorde, 2025-03-10:
I noticed some failures in:

 - ros2node
```bash
======================================================================
FAIL: test_info_node[rmw_zenoh_cpp] (test_cli.TestROS2NodeCLI.test_info_node[rmw_zenoh_cpp])
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/ahcorde/ros2_rolling/install/lib/python3.12/site-packages/launch_testing/markers.py", line 61, in _wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ahcorde/ros2_rolling/src/ros2/ros2cli/ros2node/test/test_cli.py", line 169, in test_info_node
    assert launch_testing.tools.expect_output(
AssertionError: Output does not match:
  /complex_node
    Subscribers:
      /strings: test_msgs/msg/Strings
    Publishers:
      /rosout: rcl_interfaces/msg/Log
      /parameter_events: rcl_interfaces/msg/ParameterEvent
      /arrays: test_msgs/msg/Arrays
    Service Servers:
      /complex_node/describe_parameters: rcl_interfaces/srv/DescribeParameters
      /complex_node/get_parameters: rcl_interfaces/srv/GetParameters
      /complex_node/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
      /complex_node/list_parameters: rcl_interfaces/srv/ListParameters
      /complex_node/set_parameters: rcl_interfaces/srv/SetParameters
      /complex_node/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
      /complex_node/get_type_description: type_description_interfaces/srv/GetTypeDescription
      /basic: test_msgs/srv/BasicTypes
    Service Clients:
  
    Action Servers:
      /fibonacci: test_msgs/action/Fibonacci
    Action Clients:
  
  
assert False
 +  where False = <function expect_output at 0x7bed35d7ed40>(expected_lines=<itertools.chain object at 0x7bed30719600>, text='/complex_node\n  Subscribers:\n    /strings: test_msgs/msg/Strings\n  Publishers:\n    /rosout: rcl_interfaces/msg/Lo...BasicTypes\n  Service Clients:\n\n  Action Servers:\n    /fibonacci: test_msgs/action/Fibonacci\n  Action Clients:\n\n', strict=False)
 +    where <function expect_output at 0x7bed35d7ed40> = <module 'launch_testing.tools' from '/home/ahcorde/ros2_rolling/install/lib/python3.12/site-packages/launch_testing/tools/__init__.py'>.expect_output
 +      where <module 'launch_testing.tools' from '/home/ahcorde/ros2_rolling/install/lib/python3.12/site-packages/launch_testing/tools/__init__.py'> = launch_testing.tools
 +    and   <itertools.chain object at 0x7bed30719600> = <class 'itertools.chain'>(['/complex_node', '  Subscribers:', '    /strings: test_msgs/msg/Strings', '  Publishers:', '    /arrays: test_msgs/msg/Arrays', '    /parameter_events: rcl_interfaces/msg/ParameterEvent', ...], repeat(re.compile('\\s*/complex_node/.*parameter.*: rcl_interfaces/srv/.*Parameter.*'), 0), ['  Service Clients:', '', '  Action Servers:', '    /fibonacci: test_msgs/action/Fibonacci', '  Action Clients:', ''])
 +      where <class 'itertools.chain'> = itertools.chain
 +      and   repeat(re.compile('\\s*/complex_node/.*parameter.*: rcl_interfaces/srv/.*Parameter.*'), 0) = <class 'itertools.repeat'>(re.compile('\\s*/complex_node/.*parameter.*: rcl_interfaces/srv/.*Parameter.*'), 6)
 +        where <class 'itertools.repeat'> = itertools.repeat
 +        and   re.compile('\\s*/complex_node/.*parameter.*: rcl_interfaces/srv/.*Parameter.*') = <function compile at 0x7bed3663a200>('\\s*/complex_node/.*parameter.*: rcl_interfaces/srv/.*Parameter.*')
 +          where <function compile at 0x7bed3663a200> = re.compile
 +    and   '/complex_node\n  Subscribers:\n    /strings: test_msgs/msg/Strings\n  Publishers:\n    /rosout: rcl_interfaces/msg/Lo...BasicTypes\n  Service Clients:\n\n  Action Servers:\n    /fibonacci: test_msgs/action/Fibonacci\n  Action Clients:\n\n' = <launch_testing.tools.process.ProcessProxy object at 0x7bed3071b800>.output
```
 - ros2service
```
======================================================================
FAIL: test_list_hidden[rmw_zenoh_cpp] (test_cli.TestROS2ServiceCLI.test_list_hidden[rmw_zenoh_cpp])
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/ahcorde/ros2_rolling/install/lib/python3.12/site-packages/launch_testing/markers.py", line 61, in _wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ahcorde/ros2_rolling/src/ros2/ros2cli/ros2service/test/test_cli.py", line 178, in test_list_hidden
    assert launch_testing.tools.expect_output(
AssertionError: assert False
 +  where False = <function expect_output at 0x7fb25807ed40>(expected_lines=<itertools.chain object at 0x7fb24d916170>, text='/my_ns/_hidden_echo_server/describe_parameters\n/my_ns/echo_server/describe_parameters\n/my_ns/_hidden_echo_server/ge...n/my_ns/_hidden_echo_server/get_type_description\n/my_ns/echo_server/get_type_description\n/my_ns/_echo\n/my_ns/echo\n', strict=True)
 +    where <function expect_output at 0x7fb25807ed40> = <module 'launch_testing.tools' from '/home/ahcorde/ros2_rolling/install/lib/python3.12/site-packages/launch_testing/tools/__init__.py'>.expect_output
 +      where <module 'launch_testing.tools' from '/home/ahcorde/ros2_rolling/install/lib/python3.12/site-packages/launch_testing/tools/__init__.py'> = launch_testing.tools
 +    and   <itertools.chain object at 0x7fb24d916170> = <class 'itertools.chain'>(['/my_ns/_echo'], ['/my_ns/_hidden_echo_server/describe_parameters', '/my_ns/_hidden_echo_server/get_parameter_types', '/my_ns/_hidden_e.../get_type_description', '/my_ns/_hidden_echo_server/list_parameters', '/my_ns/_hidden_echo_server/set_parameters', ...], ['/my_ns/echo'], ['/my_ns/echo_server/describe_parameters', '/my_ns/echo_server/get_parameter_types', '/my_ns/echo_server/get_parameter...y_ns/echo_server/get_type_description', '/my_ns/echo_server/list_parameters', '/my_ns/echo_server/set_parameters', ...])
 +      where <class 'itertools.chain'> = itertools.chain
 +      and   ['/my_ns/_hidden_echo_server/describe_parameters', '/my_ns/_hidden_echo_server/get_parameter_types', '/my_ns/_hidden_e.../get_type_description', '/my_ns/_hidden_echo_server/list_parameters', '/my_ns/_hidden_echo_server/set_parameters', ...] = <bound method TestROS2ServiceCLI.get_expected_builtin_services of <class 'test_cli.TestROS2ServiceCLI'>>('/my_ns/_hidden_echo_server')
 +        where <bound method TestROS2ServiceCLI.get_expected_builtin_services of <class 'test_cli.TestROS2ServiceCLI'>> = <test_cli.TestROS2ServiceCLI testMethod=test_list_hidden[rmw_zenoh_cpp]>.get_expected_builtin_services
 +      and   ['/my_ns/echo_server/describe_parameters', '/my_ns/echo_server/get_parameter_types', '/my_ns/echo_server/get_parameter...y_ns/echo_server/get_type_description', '/my_ns/echo_server/list_parameters', '/my_ns/echo_server/set_parameters', ...] = <bound method TestROS2ServiceCLI.get_expected_builtin_services of <class 'test_cli.TestROS2ServiceCLI'>>('/my_ns/echo_server')
 +        where <bound method TestROS2ServiceCLI.get_expected_builtin_services of <class 'test_cli.TestROS2ServiceCLI'>> = <test_cli.TestROS2ServiceCLI testMethod=test_list_hidden[rmw_zenoh_cpp]>.get_expected_builtin_services
[... 30 lines cut ...]
AssertionError: assert False
 +  where False = <function expect_output at 0x7fb25807ed40>(expected_lines=<itertools.chain object at 0x7fb24ed279d0>, text='/my_ns/echo_server/describe_parameters [rcl_interfaces/srv/DescribeParameters]\n/my_ns/echo_server/get_parameters [rc...er/get_type_description [type_description_interfaces/srv/GetTypeDescription]\n/my_ns/echo [test_msgs/srv/BasicTypes]\n', strict=True)
=============================== warnings summary ===============================
 +    where <function expect_output at 0x7fb25807ed40> = <module 'launch_testing.tools' from '/home/ahcorde/ros2_rolling/install/lib/python3.12/site-packages/launch_testing/tools/__init__.py'>.expect_output
 +      where <module 'launch_testing.tools' from '/home/ahcorde/ros2_rolling/install/lib/python3.12/site-packages/launch_testing/tools/__init__.py'> = launch_testing.tools
 +    and   <itertools.chain object at 0x7fb24ed279d0> = <class 'itertools.chain'>(['/my_ns/echo [test_msgs/srv/BasicTypes]'], ['/my_ns/echo_server/describe_parameters [rcl_interfaces/srv/DescribeParameters]', '/my_ns/echo_server/get_parameter_t...ters [rcl_interfaces/srv/ListParameters]', '/my_ns/echo_server/set_parameters [rcl_interfaces/srv/SetParameters]', ...])
[... cut ...] 27.0.0.1']\ninet6        : ['::1']\nnetmask  ...bscriber count    : 1\ntopic               : /incompatible_chatter\npublisher count     : 1\nsubscriber count    : 1\n"
 +  where "\n   NETWORK CONFIGURATION\ninet         : 127.0.0.1\ninet4        : ['127.0.0.1']\ninet6        : ['::1']\nnetmask  ...bscriber count    : 1\ntopic               : /incompatible_chatter\npublisher count     : 1\nsubscriber count    : 1\n" = <launch_testing.tools.process.ProcessProxy object at 0x7668434840e0>.output
```

ahcorde, 2025-03-12:
- ros2topic
```bash
test/test_bw_delay_hz.py F                                               [ 10%]
test/test_echo_pub.py F                                                  [ 40%]
test/test_use_sim_time.py F                                              [ 90%]
```

YuanYuYuan, 2025-03-13:
Results of ros2topic/test_cli.py

- test_filtered_topic_hz          
- test_list_with_verbose          
- test_multiple_topics_hz         
- test_topic_bw                   
- test_topic_endpoint_info        
- test_topic_endpoint_info_verbose
- test_topic_hz

ahcorde, 2025-03-14:
This PR https://github.com/ros2/ros2cli/pull/988 should fix issues in ros2topic test

ahcorde, 2025-03-20:
all PRs are merged

closed 2025-03-20 by ahcorde