ros2/rmw_zenoh #524: Default Router Config JSON Error

WestJeremy, 2025-03-10:
I am attempting to convert my ROS2 project to use zenoh but ran into an issue testing with the DEFAULT_RMW_ZENOH_ROUTER_CONFIG.json5. When attempting to start a router in ros2 jazzy, an error occurs on start up. 

OS Ubuntu 22.04.4

To recreate:
Start terminal with ROS2 jazzy in ws. 

sudo apt update && sudo apt install ros-jazzy-rmw-zenoh-cpp 
rm -rf build/ install/ log/
source /opt/ros/jazzy/setup.bash
pkill -9 -f ros && ros2 daemon stop
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
colcon build --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo
source install/setup.bash
ros2 run rmw_zenoh_cpp rmw_zenohd 

yeilds 

ERROR ThreadId(01) zenohc::config: Failed to read config from /opt/ros/jazzy/share/rmw_zenoh_cpp/config/DEFAULT_RMW_ZENOH_ROUTER_CONFIG.json5: JSON error: unknown field `autoconnect_strategy`, expected one of `enabled`, `address`, `interface`, `ttl`, `autoconnect`, `listen` at /home/buildfarm/.cargo/git/checkouts/zenoh-cc237f2570fab813/e4ea6f0/commons/zenoh-config/src/lib.rs:845.
logging system isn't initialized: call to rcutils_logging_console_output_handler failed.
[ros2run]: Process exited with failure 1

This may just be user error. If there are any up to date tutorials or documentation available, that would be fantastic.

Yadunund, 2025-03-10:
What workspace are you building with `colcon build`? You cannot source source /opt/ros/jazzy/setup.bash
 with Jazzy binaries of rmw_zenoh installed, and build a workspace with `rmw_zenoh` cloned in. There will be a conflict in versions of rmw_zenoh binaries and source.

If you want to use binaries, don't build rmw_zenoh from source using colcon. And if you want to use source install, make sure to purge rmw_zenoh binaries from your system and build from source in a clean terminal.

If you had previously built rmw_zenoh from source and want to rely on binaries, remove rmw_zenoh from your workspace, and build the workspace from a clean terminal.

JEnoch, 2025-03-11:
Your log shows that the error occurs with commit `e4ea6f0` of Zenoh which was introduced by https://github.com/ros2/rmw_zenoh/pull/446 and used in `rmw_zenoh` jazzy [v0.2.1](https://github.com/ros2/rmw_zenoh/compare/0.2.0...0.2.1)

The `autoconnect_strategy` config has been added to Zenoh in https://github.com/eclipse-zenoh/zenoh/pull/1742 and `rmw_zenoh` supports it since https://github.com/ros2/rmw_zenoh/pull/456 (see the addition to `DEFAULT_RMW_ZENOH_ROUTER_CONFIG.json5` in this PR).
In jazzy branch, this PR has been released with [v0.2.2](https://github.com/ros2/rmw_zenoh/releases/tag/0.2.2)

So the issue is that somehow the `ros-jazzy-rmw-zenoh-cpp` was installed with version 0.2.1, while the `DEFAULT_RMW_ZENOH_ROUTER_CONFIG.json5` config is for version 0.2.2.

WestJeremy, 2025-03-11:
Wow this was incredible help! I got everything up and running and communication working across machines on the network. To reiterate what happened in ros2 beginner language:

 I am using a docker container with jazzy and working on multiple packages in a development environment. I used the apt install method to download zenoh binaries to my workspace but this resulted in errors as the DEFAULT_RMW_ZENOH_ROUTER_CONFIG.json5 config file and ros-jazzy-rmw-zenoh-cpp were different versions.

My fix was restarting the docker container to clean it of the binaries and then I used the following commands in my workspace to create rmw_zenoh_cpp as a package that gets source installed with the rest of the ros2 environment.

cd src
git clone https://github.com/ros2/rmw_zenoh.git -b jazzy
cd ..
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro jazzy -y
export RMW_IMPLEMENTATION=rmw_zenoh_cpp

source /opt/ros/jazzy/setup.bash
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
source install/setup.bash

export ZENOH_ROUTER_CONFIG_URI=/home/ws/src/rmw_zenoh/rmw_zenoh_cpp/config/routerconfig.json5
ros2 run rmw_zenoh_cpp rmw_zenohd

closed 2025-03-11 by WestJeremy