ros2/rmw_zenoh #532: Unexpected crash in lifecycle_manager

evshary, 2025-03-12:
While running navigation2 test, a crash on lifecycle_manager occurs from time to time.
It doesn't happen only on a certain lifecycle node, so I believe this might be related to how rmw_zenoh behaves in lifecycle.
This can be easily reproduced by running `colcon test --packages-select nav2_system_tests --event-handlers console_direct+`. Several tests fail due to the same issue.

```
23: [controller_server-8] [0m[INFO 1741759927.744227888] [MPPIController]: Cleaned up MPPI Controller: FollowPath (cleanup() at /workspaces/zenoh_rmw_eval/ros2_navigation2_tests_ws/src/navigation2/nav2_mppi_controller/src/controller.cpp:56)[0m
23: [controller_server-8] [0m[INFO 1741759927.744649175] [local_costmap.local_costmap]: Cleaning up (on_cleanup() at /workspaces/zenoh_rmw_eval/ros2_navigation2_tests_ws/src/navigation2/nav2_costmap_2d/src/costmap_2d_ros.cpp:361)[0m
23: [lifecycle_manager-17] terminate called after throwing an instance of 'std::system_error'
23: [lifecycle_manager-17]   what():  Invalid argument
23: [ERROR] [lifecycle_manager-17]: process has died [pid 83001, exit code -6, cmd '/workspaces/zenoh_rmw_eval/ros2_navigation2_tests_ws/install/nav2_lifecycle_manager/lib/nav2_lifecycle_manager/lifecycle_manager --ros-args --log-level info --ros-args -r __node:=lifecycle_manager_navigation -r __ns:=/ -p use_sim_time:=True --params-file /tmp/launch_params_il_nv5on --params-file /tmp/launch_params_y4o3cp9n'].
```

evshary, 2025-04-01:
The issue happened here
https://github.com/ros/bond_core/blob/fb7ed7cbd5572c14f0176f987ac45ef7b45fc952/bondcpp/src/bond.cpp#L569
Lifecycle manager inside navigation2 uses the bond for the heartbeat. The mutex inside the Bond object is destroyed earlier than the subscriber, but the callback of the subscriber requires the mutex. It will cause panic if the subscriber receives messages and triggers the callback.

I'm not sure why this doesn't happen in other DDS implementations, but perhaps the better way is to correct the behavior in the bond library.

evshary, 2025-04-02:
Might be related to the discussion https://github.com/ros/bond_core/pull/108

ahcorde, 2025-04-16:
PR is merged,closing this one for now

closed 2025-04-16 by ahcorde