ros2/rmw_zenoh #562: `RMW_ZENOH_LOG` doesn't work in the stage of config generation

YuanYuYuan, 2025-03-20:
1. Set the override to a invalid value.
```bash
export ZENOH_CONFIG_OVERRIDE='yeeeeeeeeeeeeeee=3000'
```
2. Run the rmw_zenohd
```bash
➜ ros2 run rmw_zenoh_cpp rmw_zenohd
2025-03-20T09:06:32.457268Z ERROR ThreadId(01) zenohcd::config: Failed to insert value '3000' for key 'yeeeeeeeeeeeeeeeeeee' into config: Str("unknown key") at /home/ubuntu/.cargo/git/checkouts/zenoh-cc237f2570fab813/b661454/zenoh/src/api/config.rs:67.
logging system isn't initialized: call to rcutils_logging_console_output_handler failed.
2025-03-20T09:06:32.457607Z  INFO ThreadId(01) zenoh::net::runtime: Using ZID: 3209190b3b8a3f9942b9250eff37b917
2025-03-20T09:06:32.460416Z  INFO ThreadId(01) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fe80::1a8f:9f5b:336a:3eb0]:7447
2025-03-20T09:06:32.460515Z  INFO ThreadId(01) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.68.54:7447
Started Zenoh router with id3209190b3b8a3f9942b9250eff37b917
^C2025-03-20T09:06:36.530327Z  INFO ThreadId(01) zenoh::api::session: close session zid=3209190b3b8a3f9942b9250eff37b917
```

This means [this line](https://github.com/ros2/rmw_zenoh/blob/20ee1ceaedbd22701e2686696c088efdcebec6d8/rmw_zenoh_cpp/src/detail/zenoh_config.cpp?plain=1#L95) of logging doesn't work.

Yadunund, 2025-03-20:
hmm I tried this and it works

```bash
yadunund@ubuntu-24-04:~/ros2_rolling_ci$ ZENOH_CONFIG_OVERRIDE='foo=bar' ros2 run demo_nodes_cpp talker
2025-03-20T20:36:51.771728Z ERROR ThreadId(01) zenohc::config: Failed to insert value 'bar' for key 'foo' into config: Json5Err(Message { msg: " --> 1:1\n  |\n1 | bar\n  | ^---\n  |\n  = expected array, boolean, null, number, object, or string", location: Some(Location { line: 1, column: 1 }) }) at /home/yadunund/.cargo/git/checkouts/zenoh-cc237f2570fab813/b661454/zenoh/src/api/config.rs:67.
[WARN] [1742503011.771776036] [rmw_zenoh_cpp]: Ignore the invalid configuration key-value pair: (foo, bar)
2025-03-20T20:36:52.274038Z  WARN ThreadId(01) zenoh::net::runtime::orchestrator: Scouting delay elapsed before start conditions are met.
[INFO] [1742503013.282788991] [talker]: Publishing: 'Hello World: 1'
[INFO] [1742503014.282778316] [talker]: Publishing: 'Hello World: 2'
```

YuanYuYuan, 2025-03-21:
Okay. So apparently the issue occurs at rmw_zenohd.

```bash
➜ ZENOH_CONFIG_OVERRIDE='foo=bar' ros2 run demo_nodes_cpp talker
2025-03-21T03:25:02.238000Z ERROR ThreadId(01) zenohcd::config: Failed to insert value 'bar' for key 'foo' into config: Json5Err(Message { msg: " --> 1:1\n  |\n1 | bar\n  | ^---\n  |\n  = expected array, boolean, null, number, object, or string", location: Some(Location { line: 1, column: 1 }) }) at /home/ubuntu/.cargo/git/checkouts/zenoh-cc237f2570fab813/b661454/zenoh/src/api/config.rs:67.
[WARN] [1742527502.238051739] [rmw_zenoh_cpp]: Ignore the invalid configuration key-value pair: (foo, bar)
2025-03-21T03:25:02.238332Z  INFO ThreadId(01) zenoh::net::runtime: Using ZID: afa358877aca5b6bc05c0ccdc124ca70
2025-03-21T03:25:02.248305Z  INFO ThreadId(01) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[::1]:41327
2025-03-21T03:25:02.750264Z  WARN ThreadId(01) zenoh::net::runtime::orchestrator: Scouting delay elapsed before start conditions are met.
[WARN] [1742527502.750398576] [rmw_zenoh_cpp]: Unable to connect to a Zenoh router. Have you started a router with `ros2 run rmw_zenoh_cpp rmw_zenohd`?
[WARN] [1742527502.750417892] [rmw_zenoh_cpp]: Unable to connect to a Zenoh router after 1 attempt(s). Please ensure that a Zenoh router is running and can be reached. You may increase the number of attempts to check for a router by setting the ZENOH_ROUTER_CHECK_ATTEMPTS environment variable. Proceeding with initialization but other peers will not discover or receive data from peers in this session until a router is started.
^C[INFO] [1742527503.212795822] [rclcpp]: signal_handler(signum=2)
2025-03-21T03:25:03.217502Z  INFO ThreadId(01) zenoh::api::session: close session zid=afa358877aca5b6bc05c0ccdc124ca70

➜ ZENOH_CONFIG_OVERRIDE='foo=bar' ros2 run rmw_zenoh_cpp rmw_zenohd
2025-03-21T03:25:16.870820Z ERROR ThreadId(01) zenohcd::config: Failed to insert value 'bar' for key 'foo' into config: Json5Err(Message { msg: " --> 1:1\n  |\n1 | bar\n  | ^---\n  |\n  = expected array, boolean, null, number, object, or string", location: Some(Location { line: 1, column: 1 }) }) at /home/ubuntu/.cargo/git/checkouts/zenoh-cc237f2570fab813/b661454/zenoh/src/api/config.rs:67.
logging system isn't initialized: call to rcutils_logging_console_output_handler failed.
2025-03-21T03:25:16.871085Z  INFO ThreadId(01) zenoh::net::runtime: Using ZID: c5dfbf534a4fe75105005c9dd80ed941
2025-03-21T03:25:16.874868Z  INFO ThreadId(01) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fe80::1a8f:9f5b:336a:3eb0]:7447
2025-03-21T03:25:16.875135Z  INFO ThreadId(01) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fe80::216:3eff:fe00:1]:7447
2025-03-21T03:25:16.875163Z  INFO ThreadId(01) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.68.54:7447
2025-03-21T03:25:16.875169Z  INFO ThreadId(01) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.240.1:7447
Started Zenoh router with idc5dfbf534a4fe75105005c9dd80ed941
^C2025-03-21T03:25:17.528613Z  INFO ThreadId(01) zenoh::api::session: close session zid=c5dfbf534a4fe75105005c9dd80ed941
```