ros2/rmw_zenoh #563: Pass all the tests in P2P mode

YuanYuYuan, 2025-03-20:
By setting 
```
export ZENOH_CONFIG_OVERRIDE=scouting/multicast/enabled=true;scouting/multicast/address="224.0.0.224:<PORT>"
```
we can run the `ros2.repos` tests without the rmw_zenohd.

## Test failures

```
  launch_testing_ros
  ros2cli
  rosbag2_tests
  ros2lifecycle
  sros2
  test_tracetools
  ros2service
  ros2node
  ros2topic
```

YuanYuYuan, 2025-03-20:
We need to disable the `RUST_LOG` since the warnings pop out and fail some python tests.

```
export RUST_LOG=z=error
```

## Test Failures

```
sros2
sros2
```

Yadunund, 2025-03-20:
@YuanYuYuan if we set <Port> to 0, will Zenoh automatically pick and available port?

Yadunund, 2025-03-20:
Test failures without enabling muticast as reported from a CI job: https://ci.ros2.org/job/ci_linux-aarch64/17033/#showFailuresLink

Yadunund, 2025-03-21:
@YuanYuYuan I've taken your findings from here and created https://github.com/ros2/rmw_zenoh/issues/567. let's check off items on the list there as we submit PRs to the core repos.