ros2/rmw_zenoh #567: Get ci.ros2.org to pass with rmw_zenoh

Yadunund, 2025-03-21:
After a huge effort in https://github.com/ros2/rmw_zenoh/issues/493, we have all tests for all packages in the ros2.repos file passing with rmw_zenoh in local workspaces 🎉 

However this required us to start the zenohd router locally which does not happen automatically when running a CI job on ci.ros2.org. The ros-infrastructure team is working on middleware agnostic mechanisms to start a pre-defined process but this may not be ready before the RMW freeze for Kilted on April 7. 

A workaround in the meanwhile (to demonstrate that rmw_zenoh is indeed capable of passing CI) is to have ci.ros2.org run the usual tests with multicast scouting enabled for discovery. This would be akin to DDS-based middlewares. Once the infra tooling is in place, we can revert these changes.

The good news is that while we already have all tests for all packages passing locally, when a job runs on ci.ros2.org, tests for most packages run with the default middleware and only a select few run in ros2.repos file generate tests for every middleware found in the workspace. These packages rely on the `get_available_rmw_implementations` macro to retrieve all middlewares built and generate tests for these middlewares. Therefore the goal is to update the tests in these repos to set environment variables to run rmw_zenoh tests with multicast discovery enabled. ie, set 

```
export ZENOH_ROUTER_CHECK_ATTEMPTS=-1
export ZENOH_CONFIG_OVERRIDE='scouting/multicast/enabled=true'
export RUST_LOG=z=error
```

The packages that require changes are
- [ ] test_tracetools
- [ ] rclpy
- [ ] ros2param
- [ ] ros2service
- [ ] ros2topic
- [ ] ros2doctor
- [ ] ros2node
- [ ] ros2lifecycle
- [ ] ros2action
- [ ] pendulum_control
- [ ] test_quality_of_service
- [ ] test_communication
- [ ] test_security
- [ ] test_rclcpp
- [ ] sros2
- [ ] rosbag2_cpp

YuanYuYuan, 2025-03-21:
Hi @Yadunund! Since the previous issue has been closed, let me answer you here. We can set 
```
export ZENOH_CONFIG_OVERRIDE='scouting/multicast/enabled=true;scouting/multicast/address="224.0.0.224:0"'
```
to for port auto selection, but more tests fail compared to generating the ports in a script.

YuanYuYuan, 2025-03-21:
Regarding the failures reported by CI, I suspect the rmw_zenohd router doesn't work well. See the log [here](https://ci.ros2.org/job/ci_linux/22803/testReport/junit/(root)/projectroot/test_n_nodes__rmw_zenoh_cpp/).

Yadunund, 2025-03-21:
> Regarding the failures reported by CI, I suspect the rmw_zenohd router doesn't work well. See the log [here](https://ci.ros2.org/job/ci_linux/22803/testReport/junit/(root)/projectroot/test_n_nodes__rmw_zenoh_cpp/).

@YuanYuYuan to clarify, all these tests are failing because the Zenoh router has not been started before running the tests. Hence we need to add changes like https://github.com/ros2/rcl/pull/1218 to these packages.

YuanYuYuan, 2025-03-21:
> [@YuanYuYuan](https://github.com/YuanYuYuan) to clarify, all these tests are failing because the Zenoh router has not been started before running the tests. Hence we need to add changes like [ros2/rcl#1218](https://github.com/ros2/rcl/pull/1218) to these packages.

I see. Then let's focus on the multicast version first. Here are the tests that haven't steadily passed on a local Ubuntu.

```
╭───┬────────────────────╮
│ 0 │ ros2param          │
│ 1 │ lttngpy            │
│ 2 │ rosbag2_tests      │
│ 3 │ launch_testing_ros │
│ 4 │ ros2doctor         │
│ 5 │ ros2lifecycle      │
│ 6 │ ros2topic          │
╰───┴────────────────────╯
```

Yadunund, 2025-03-22:
We are down to 1 test failure on Linux/RHEL and 75 on Windows.

* Linux [image]
* Linux-aarch64 [image]
* Linux-rhel [image]
* Windows [image]

YuanYuYuan, 2025-03-24:
Remaining test failures:

- rosbag2/test_record_qos_profiles: Windows-only https://github.com/ros2/rosbag2/issues/1947
- sros2/test_generate_policy: flaky
- test_launch_ros/pytest.missing_result

Yadunund, 2025-03-24:
CI started with: https://raw.githubusercontent.com/ros2/ros2/3f873e9c5a79ce3b47f70199708b21c07007cd7d/ros2.repos
 
* Linux [image]
* Linux-aarch64 [image]
* Linux-rhel [image]
* Windows [image]

Yadunund, 2025-03-26:
CI after rosbag2 windows fix was merged 

* Linux [image]
* Linux-aarch64 [image]
* Linux-rhel [image]
* Windows [image]

ahcorde, 2025-03-31:
Thank you everyone who help to get the CI green with Zenoh!

closed 2025-03-31 by ahcorde