ros2/rmw_zenoh #568: Rolling [rcutils|error_handling.c:108] rcutils_set_error_state()

berndpfrommer, 2025-03-22:
After upgrading to the latest ``rmw-zenoh-cpp`` package on Rolling I get this warning message the first time I run e.g. ``ros2 topic list``:
```
 ros2 topic list

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'failed to resolve symbol 'rmw_event_type_is_supported' in shared library '/opt/ros/rolling/lib/librmw_zenoh_cpp.so', at ./src/functions.cpp:152'

with this new error message:

  'service's implementation is invalid, at ./src/rcl/service.c:428'

rcutils_reset_error() should be called after error handling to avoid this.
```

On subsequent calls I don't get the error message again, probably some caching by the ros2 daemon. Stopping the ros2 daemon will cause the warning message to reappear.

Installed package:
```
ros-rolling-rmw-zenoh-cpp/noble,now 0.4.0-1noble.20250314.035117 amd64
```

Yadunund, 2025-03-22:
Is your version of `rmw` pakcage 7.8.1? If so this should be resolved when rmw_zenoh 0.5.0 is out https://github.com/ros/rosdistro/pull/44848 (should be by the end of next week).

In the meanwhile you may build `rmw_zenoh` from source to avoid the error.

berndpfrommer, 2025-03-23:
This is what apt-search tells me:
```
ros-rolling-rmw-zenoh-cpp/noble,now 0.4.0-1noble.20250314.035117
```

Yadunund, 2025-04-03:
Should be resolved with  an `apt update && apt upgrade` since the Rolling sync [went out last week](https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2025-03-28/42863). Feel free to reopen if you still have the issue.

closed 2025-04-03 by Yadunund

berndpfrommer, 2025-04-04:
confirmed works for me now. Thanks!