ros2/rmw_zenoh #571: "Received liveliness token with invalid qos keyexpr"

robin-zealrobotics, 2025-03-23:
When using rmw_zenoh, pretty much every node gives the error
` [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr`
I'm not sure if I'm forgetting to configure something or did something wrong.

I'm running rmw_zenohd in the same launch file as my other nodes, but 1 second before all nodes. I'm running jazzy on noble in a docker container.

Yadunund, 2025-03-23:
Please provide more information about your setup including
- How ROS 2 Jazzy is installed (binaries vs source)
- How rmw_zenoh is installed and what version (binary vs source)
- A minimal reproducible example of the issue

robin-zealrobotics, 2025-03-24:
Jazzy with binaries and rmw_zenoh with binaries (0.2.2-1noble.20250220.095611)

I've pinned the problem down to one very particular 'spawner' node from ros2_control. My launch has 4 spawner nodes. If I keep my launch as is, but leave out the one problematic spawner, I see no errors anywhere.

Below's a snippet of the log. The problematic node is spawner-5 (starting odometry_broadcaster), but it results in errors to many other nodes as well.

I'm still investigating a bit further, I'll let you know if I can pin it down to some particular configuration.

```
[ros2_control_node-3] [INFO] [2025-03-24 10:02:04.501] [runner_0.controller_manager]: Controller 'odometry_broadcaster' node arguments: --ros-args --params-file /ament/overlay_ws/install/share/zeal_launch/config/vehicles/zeal_runner.config.yaml -p use_sim_time:=False --params-file /ament/overlay_ws/install/share/zeal_launch/config/motion/ros2_control.config.yaml -r /diagnostics:=/runner_0/diagnostics -r /parameter_events:=/runner_0/parameter_events --ros-args --param odom_frame_id:=runner_0/odom --param base_frame_id:=runner_0/base_link --param enable_odom_tf:=true --remap odometry_broadcaster/odom:=localization/odom 
[spawner-5] [INFO] [2025-03-24 10:02:04.512] [runner_0.spawner_odometry_broadcaster]: Loaded odometry_broadcaster
[ros2_control_node-3] [INFO] [2025-03-24 10:02:04.513] [runner_0.controller_manager]: Configuring controller: 'odometry_broadcaster'
[ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[ros2_control_node-3] [ERROR] [2025-03-24 10:02:04.514] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[wait_server-12] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[mock_field_select-18] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[ERROR] [2025-03-24 10:02:04.516] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[kinematic_profile_aggregator-9] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[lost_limiter-13] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[spawner-5] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[mode_manager-14] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[mock_low_level_bridge-17] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[path_follower-10] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[mock_ekf-19] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[field_set_publisher-15] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[robot_state_publisher-2] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[collision_detector_node-22] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[move_action_server-11] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[twist_mux-8] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[point_cloud_merger-21] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[spawner-6] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[path_planner-16] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[component_container-23] [ERROR] [2025-03-24 10:02:04.515] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[ros2_control_node-3] 
[ros2_control_node-3] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[ros2_control_node-3] This error state is being overwritten:
[ros2_control_node-3] 
[ros2_control_node-3]   'Error setting QoS values from strings: unordered_map::at, at ./src/detail/liveliness_utils.cpp:340'
[ros2_control_node-3] 
[ros2_control_node-3] with this new error message:
[ros2_control_node-3] 
[ros2_control_node-3]   'Handle's typesupport identifier (rosidl_typesupport_cpp) is not supported by this library, at ./src/type_support_dispatch.hpp:114'
[ros2_control_node-3] 
[ros2_control_node-3] rcutils_reset_error() should be called after error handling to avoid this.
[ros2_control_node-3] <<<
[ros2_control_node-3] [ERROR] [2025-03-24 10:02:04.516] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[mock_ekf-19] [ERROR] [2025-03-24 10:02:04.516] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[component_container-23] [ERROR] [2025-03-24 10:02:04.516] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[robot_state_publisher-2] [ERROR] [2025-03-24 10:02:04.516] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[path_follower-10] [ERROR] [2025-03-24 10:02:04.516] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[twist_mux-8] [ERROR] [2025-03-24 10:02:04.517] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[... cut ...]
[wait_server-12] [ERROR] [2025-03-24 10:02:04.517] [rmw_zenoh_cpp]: Received liveliness token with invalid qos keyexpr
[ros2_control_node-3] [INFO] [2025-03-24 10:02:04.520] [runner_0.controller_manager]: Activating controllers: [ odometry_broadcaster ]
[INFO] [spawner-7]: process has finished cleanly [pid 40368]
[spawner-5] [INFO] [2025-03-24 10:02:04.528] [runner_0.spawner_odometry_broadcaster]: Configured and activated odometry_broadcaster
[INFO] [spawner-5]: process has finished cleanly [pid 40366]
```

robin-zealrobotics, 2025-03-24:
Found it faster than anticipated. The culprit is this line in a custom ros2_control controller:
```
_pub_odometry = get_node()->create_publisher<nav_msgs::msg::Odometry>(
    DEFAULT_ODOMETRY_TOPIC, rclcpp::BestAvailableQoS());
```
This work in cyclone, but doesn't in zenoh.
If I change it to e.g.
```
_pub_odometry = get_node()->create_publisher<nav_msgs::msg::Odometry>(
    DEFAULT_ODOMETRY_TOPIC, rclcpp::SystemDefaultsQoS());
```
All errors disappear. So this single line in one node is triggering the error message on pretty much every node in the launch file.

Yadunund, 2025-03-24:
The binaries are bit outdated so please try building the latest source version.

Specifically we changed the SystemDefault durability to `volatile` (from `transient_local`) which matches other DDS implementations. This change has already been bloomed https://github.com/ros/rosdistro/pull/44849 and will be available after the next Jazzy sync.

Yadunund, 2025-04-05:
Closing as this issue should be resolved but feel free to reopen if otherwise.

closed 2025-04-05 by Yadunund