ros2/rmw_zenoh #580: ROS2 Topic list not showing zenoh-pico nodes

PetervdPerk-NXP, 2025-03-27:
### Operating System:

Linux and NuttX

### ROS version or commit hash:

Jazzy

### 'ros2 doctor --report' output

```Formatted

```

### Steps to reproduce issue

I've got a Zenoh-pico node that's interoperable with ROS2.
https://github.com/PX4/PX4-Autopilot/pull/24476

When I type `ros2 topic list` I don't see the topics though.
Whereas manually echoing it's working `ros2 topic echo sensor px4_msgs/SensorCombined`

I see on wireshark that my zenoh-pico node declares the topic with required information
- Domain id
- Topic name
- Type name
- Typehash

[image]

So I presume that zenohd has knowledge that there's a publisher ready to publish this type of data but it's not expose to `ros2 topic list`

### Expected behavior

When typing I would expect my topic show up i.e.
```
ros2 topic list
/rosout
/parameter_events
/sensor (this one is not showing up)
```

### Actual behavior

```
ros2 topic list
/rosout
/parameter_events
```

JEnoch, 2025-03-28:
the `ros2` command relies on the ROS graph to show all nodes/topics/services/actions lists. The Zenoh's declarations messages such as the one you see sent by zenoh-pico are not sufficient to discover all the information required by a ROS graph. Therefore `rmw_zenoh` relies on extra Zenoh liveliness tokens created by each Nodes to declare and monitor the ROS graph.

For your zenoh-pico node to participate to the ROS graph, it has to declare liveliness tokens with key expressions in the way `rmw_zenoh` expects them. See here how this `liveliness_keyexpr_` is constructed:
https://github.com/ros2/rmw_zenoh/blob/948753ab26492202f11b34a808e3bdd8ed2702f4/rmw_zenoh_cpp/src/detail/liveliness_utils.cpp#L398-L433 

Some examples of such key expressions:
- Node (`NN`): `@ros2_lv/0/aa10815c9800f839bd87d6bf719b6ac7/0/0/NN/%/%/talker`
- Publisher (`MP`): `@ros2_lv/0/aa10815c9800f839bd87d6bf719b6ac7/0/11/MP/%/%/talker/%chatter/std_msgs::msg::dds_::String_/RIHS01_df668c740482bbd48fb39d76a70dfd4bd59db1288021743503259e948f6b1a18/::,7:,:,:,,`

And here you have an example of liveliness token declaration with zenoh-pico:
https://github.com/eclipse-zenoh/zenoh-pico/blob/main/examples/unix/c11/z_liveliness.c

PetervdPerk-NXP, 2025-03-28:
Awesome that's what I was looking for I've got the liveliness to work now.

```
peter@desktop-peter:~/ros2_ws$ ros2 node info /px4_aabbcc00000000000000000000000000
/px4_aabbcc00000000000000000000000000
  Subscribers:
    /test_topic: px4_msgs/msg/SensorCombined
  Publishers:
    /vehicle_local_position: px4_msgs/msg/VehicleLocalPosition
    /sensor: px4_msgs/msg/SensorCombined
    /distance_sensor: px4_msgs/msg/DistanceSensor
    /vehicle_global_position_groundtruth: px4_msgs/msg/VehicleGlobalPosition
    /vehicle_imu: px4_msgs/msg/VehicleImu
    /vehicle_angular_velocity_groundtruth: px4_msgs/msg/VehicleAngularVelocity
    /cpuload: px4_msgs/msg/Cpuload
    /vehicle_control_mode: px4_msgs/msg/VehicleControlMode
    /vehicle_attitude: px4_msgs/msg/VehicleAttitude
    /vehicle_attitude_groundtruth: px4_msgs/msg/VehicleAttitude
    /estimator_aid_src_fake_pos: px4_msgs/msg/EstimatorAidSource2d
    /estimator_aid_src_gravity: px4_msgs/msg/EstimatorAidSource3d
    /vehicle_imu_status: px4_msgs/msg/VehicleImuStatus
    /vehicle_odometry: px4_msgs/msg/VehicleOdometry
    /vehicle_acceleration: px4_msgs/msg/VehicleAcceleration
    /actuator_armed: px4_msgs/msg/ActuatorArmed
    /battery_status: px4_msgs/msg/BatteryStatus
    /vehicle_local_position_groundtruth: px4_msgs/msg/VehicleLocalPosition
    /estimator_aid_src_rng_hgt: px4_msgs/msg/EstimatorAidSource1d
    /vehicle_angular_velocity: px4_msgs/msg/VehicleAngularVelocity
  Service Servers:

  Service Clients:

  Action Servers:

  Action Clients:

peter@desktop-peter:~/ros2_ws$
```

closed 2025-03-28 by PetervdPerk-NXP

Yadunund, 2025-03-29:
@PetervdPerk-NXP it would be great to document your findings in our README.md. Perhaps adding a section before the `Known Issues` section on communicating between zenoh-pico and rmw_zenoh. Would you be willing to open a PR?

PetervdPerk-NXP, 2025-03-29:
Hi @Yadunund ,
I'm currently re-implementing the functionality of `rmw_zenoh_cpp`, using `zenoh-pico` as the underlying layer for embedded usecases. While `zenoh-cpp` and `zenoh-pico` _should_ be fully compatible, the challenge lies in how `rmw_zenoh` utilizes Zenoh’s features to integrate with rmw/rclc.

The main issues I've encountered so far is trying to understand how `rmw_zenoh` handles serialization, key expression (keyexpr) definitions and liveliness. At this stage, the code is evolving faster than the documentation, which makes sense given that `rmw_zenoh` is still in rapid development and not yet stable.

Once `rmw_zenoh` stabilizes, I propose adding dedicated chapters in the design.md file to documenting how `rmw_zenoh` utilizes zenoh features such as serialization, keyexpr, liveliness and probably even more. This would serve as a reference for implementing the `rmw_zenoh` interopability using other Zenoh client libraries, such as zenoh-python, zenoh (Rust), zenoh-csharp, and zenoh-pico.

xopxe, 2025-05-02:
> Hi [@Yadunund](https://github.com/Yadunund) , I'm currently re-implementing the functionality of `rmw_zenoh_cpp`, using `zenoh-pico` as the underlying layer for embedded usecases. While `zenoh-cpp` [...]

I',m very interested in a zeno-pico-rmw implementation. We are looking for ways to integrate ESP32-class microcontrollers to ROS 2 deployments. A rationale is (here)[https://discourse.ros.org/t/integrating-ros-2-with-microcontrollers-when-using-zenoh/43463]. 

In your experience, how close is the ecosystem to providing such a thing? Is there any code that can be looked up? We would love to help out.