ros2/rmw_zenoh #623: Setting DiffServe value of select topics

gstangel, 2025-04-27:
### Description

For use cases involving command and control ROS2 streams, setting the Differentiated Services field of a IP frame is a useful feature.  For example, in drone swarms that may be connected via a smart meshed WiFi Radio, the differentiated service field can be used in times of contention to ensure that the radio selectively prioritizes command and control streams.

### Motivation

ROS2 has no inherent ability to set a 'hard' network level priority on data.  The different QoS Settings are applied at the software layer.  This doesn't necessarily make sense with a TCP Stream, but for UDP data it can be helpful

JEnoch, 2025-05-20:
Zenoh now supports DiffServ via a new `dscp` (differentiated services code point) configuration per-endpoint.
See https://github.com/eclipse-zenoh/zenoh/pull/1937 which resolves https://github.com/eclipse-zenoh/zenoh/issues/1924.

It will be available in `rmw_zenoh` with the next bump of `zenoh-c` commit id.

JEnoch, 2025-05-21:
@gstangel `zenoh-c` and `zenoh-cpp` are now synced with `zenoh` wrt. `dscp` feature.
Waiting for the bump in `rmw_zenoh`, could you test this feature in advance ?

You need to:
1. build from sources, changing the commit ids in zenoh_cpp_vendor/CMakeLists.txt with:
   - `zenoh-c`: d00d5da2226b084d83cedf0545d3976d5a630c7c
   - `zenoh-cpp`: c96600c9a91ea5a63046c3830e5e442fd255f3a1
  2. configure 2 routers across you meshed WiFi to establish 2 TCP connections with different `priority` and `dscp`, and to configure priority overwrite when routing over the WiFi interface

Here is an example of the changes to apply in the routers config (assuming one has an interface "en0" over WiFi with IP 192.168.0.2).
Router 1 (listening on 192.168.0.2 over WiFi):
```json5
{
    listen: {
        endpoints: [
            "tcp/127.0.0.1:7447",  // For the Nodes over loopback
            "tcp/192.168.0.2:8888?prio=4-7",   // Low priority over WiFi
            "tcp/192.168.0.2:9999?prio=0-3#dscp=0x48"]  // High priority over WiFi
    },
    // Allow 2 links between 2 Sessions
    transport: { unicast: { max_links: 2 } },
    // QoS overwrite per-network interface
    qos: {
        network: [
            {
                // Overwrite priority to interactive_high (=2) for some topics
                id: "router_to_router_high_prio",
                interfaces: ["en0"],
                messages: ["put", "query", "reply"],
                flows: ["egress"],
                key_exprs: ["*/topic_high_prio/*/*"],
                overwrite: { priority: "interactive_high" }
            }
        ]
    }
}
```

Router 2 (connecting to 192.168.0.2 over WiFi):
```json5
{
    connect: {
        endpoints: [
            "tcp/192.168.0.2:8888?prio=4-7",  // Low priority over WiFi
            "tcp/192.168.0.2:9999?prio=0-3#dscp=0x48"  // High priority over WiFi
        ]
    },
    listen: {
        endpoints:["tcp/127.0.0.1:7447"]  // For the Nodes over loopback
    },
    // Allow 2 links between 2 Sessions
    transport: { unicast: { max_links: 2 } },
    // QoS overwrite per-network interface
    qos: {
        network: [
            {
                id: "router_to_router_high_prio",
                interfaces: ["en0"],
                messages: ["put", "query", "reply"],
                flows: ["egress"],
                key_exprs: ["*/topic_high_prio/*/*"],
                overwrite: { priority: "interactive_high" }
            }
        ]
    }
}
```

Then you can test and check the traffic on Wireshark for instance, running:
- for low priority traffic: 
   - `ros2 topic pub /chatter std_msgs/String "data: Hello World"`
   - `ros2 topic echo /chatter std_msgs/String`
- for high priority traffic:
   - `ros2 topic pub /topic_high_prio std_msgs/String "data: Hello World"`
   - `ros2 topic echo /topic_high_prio std_msgs/String`

gstangel, 2025-05-21:
My use case is mostly oriented for zenohd.  I have my ROS2 nodes running on isolated bridges depending on their importance.  So `br_sys` has gateway 10.10.0.1 and `br_experiment` has gateway 10.20.0.1. 

I then have zenohd listening on both bridge interfaces.  Using this new DSCP functionality, if my listen is

```
    listen: {
        endpoints: [
            "tcp/127.0.0.1:7447", 
            "tcp/10.20.0.1",   // Low priority, don't set DSCP
            "tcp/10.10.0.1#dscp=0x48"]  // Listen on br_sys, set DSCP of traffic coming out of this bridge to 0x48
    },
```

Is the expected behavior that this will set the DSCP value of all traffic coming out of this bridge, and that it will be respected by Zenoh throughout the network? eg that the DSCP field will not be cleared during hops from router to router?

gstangel, 2025-05-21:
I guess I could use the network.qos option to filter based on what bridge they originated from as well.  Eg

```
qos: {
        network: [
            {
                id: "router_to_router_high_prio",
                interfaces: ["br_sys"],
                messages: ["put", "query", "reply"],
                flows: ["egress"],
                key_exprs: ["*/**"],
                overwrite: { priority: "interactive_high" }
            }
        ]
    }
```

JEnoch, 2025-05-23:
> Is the expected behavior that this will set the DSCP value of all traffic coming out of this bridge

Yes, using `"tcp/10.10.0.1#dscp=0x48"` the `IP_TOS` option with the DSCP value is set to the socket and thus to every packet sent from this socket. For IPv6 the `IPV6_TCLASS` option is set.

> and that it will be respected by Zenoh throughout the network ?

No, the DSCP flag is configured at socket creation and applied for all packets going out this socket.

> eg that the DSCP field will not be cleared during hops from router to router?

The filed will be cleared when routing. But the Zenoh priority is preserved and applied. Thus when routing to another router, if there are 2 possible endpoints configured with different priority ranges, `zenohd` will route the message via the endpoint with a matching priority.
Thus it's important that you define on your endpoints different priority ranges for the different DSCP values. 
This allows you to:
1. configure the priority depending the key expression (hence topic name)
2. let Zenoh to use the appropriate endpoint (with configured DSCP) depending on the message's priority