ros2/rmw_zenoh #624: [zenoh_cpp_vendor] Linker fails with LTO enabled (libzenohc.so)

oysstu, 2025-05-01:
### Operating System:

Linux os-home 6.12.25-2-lts #1 SMP PREEMPT_DYNAMIC Tue, 29 Apr 2025 09:49:19 +0000 x86_64 GNU/Linux

### ROS version or commit hash:

kilted beta (2025.04.30)

### RMW Configuration (if applicable):

The linker fails for libzenohc.so with the following undefined symbols with LTO enabled. This is with GCC 15.1.1. Disabling LTO seems to resolve the issue.

```
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p256_scalar_mul_mont'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p256_sqr_mont'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p256_point_mul_base'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__LIMBS_window5_split_window'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__LIMBS_less_than'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p384_elem_mul_mont'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__LIMBS_sub_mod'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p384_point_add'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__aes_nohw_ctr32_encrypt_blocks'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_ge_double_scalarmult_vartime'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p256_point_mul_base_vartime'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_public_from_private_generic_masked'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p384_point_mul'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_ge_scalarmult_base'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__LIMBS_add_mod'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_sc_mask'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__bn_neg_inv_mod_r_u64'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__bn_from_montgomery_in_place'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p256_point_mul'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__aes_nohw_encrypt'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_sc_muladd'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__OPENSSL_cpuid_setup'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_scalar_mult_generic_masked'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p256_scalar_sqr_rep_mont'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p256_point_add'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_fe_invert'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p256_mul_mont'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__LIMB_shr'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_scalar_mult_adx'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__adx_bmi2_available'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__LIMBS_window5_unsplit_window'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__LIMBS_reduce_once'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__p384_scalar_mul_mont'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__LIMB_is_zero'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_fe_mul_ttt'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_fe_isnegative'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_fe_tobytes'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_sc_reduce'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__avx2_available'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__CRYPTO_memcmp'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__LIMBS_shl_mod'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_ge_frombytes_vartime'
/usr/bin/ld: src/install/opt/zenoh_cpp_vendor/lib/libzenohc.so: undefined reference to `ring_core_0_17_13__x25519_fe_neg'
collect2: error: ld returned 1 exit status
```

### 'ros2 doctor --report' output

### Steps to reproduce issue

1. Build with GCC and -flto=auto

### Expected behavior

Build success.

### Actual behavior

Build failure.

JEnoch, 2025-05-01:
This is probably related to https://github.com/briansmith/ring/issues/1444

`ring` is a cryptography library used by both `rustls` and `quinn` libraries. Zenoh depends on those respectively for TLS and QUIC transports.
The TLS transport is built by default in `rmw_zenoh`. If you don't need it, you can remove `zenoh/transport_tls` from [zenoh_cpp_vendor/CMakeLists.txt](https://github.com/ros2/rmw_zenoh/blob/2771bc22a2646b8b37382a1379538cdd51a4c4b2/zenoh_cpp_vendor/CMakeLists.txt#L18) and try to build with LTO enabled.

oysstu, 2025-05-02:
That seems to be the case. I can confirm that it compiles with LTO after removing `zenoh/transport_tls`. 

Do you know if there's a good way to disable LTO for only this component or the libzenohc library? I'm packaging this for others so I'd like to avoid disabling features if possible.

JEnoch, 2025-05-12:
I don't think you can disable LTO only for this dependency.
You can disable LTO when building libzenohc library setting this environment variable when calling colcon:
`CARGO_BUILD_RUSTFLAGS="-C lto=false"`