ros2/rmw_zenoh #626: cannot enable multicast

christian-rauch, 2025-05-07:
### Operating System:

Ubuntu 24.04

### ROS version or commit hash:

jazzy

### RMW implementation (if applicable):

rmw_zenoh

### RMW Configuration (if applicable):

Trying to use `ZENOH_CONFIG_OVERRIDE='scouting/multicast/enabled=true'`.

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
usage: ros2 [-h] [--use-python-default-buffering]
            Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'doc' (choose from 'action', 'bag', 'component', 'control', 'daemon', 'doctor', 'extension_points', 'extensions', 'interface', 'launch', 'lifecycle', 'multicast', 'node', 'param', 'pkg', 'run', 'security', 'service', 'topic', 'wtf')
```
</details>

### Steps to reproduce issue

Follow the instructions for "Enable multicast scouting (disabled by default) that allows ROS2 nodes to discover each other without requiring a Zenoh router:".

Setup: PC A (inet 10.71.1.208/24) and B (inet 10.71.1.207/24) in the same subnet.

```sh
# A
export ZENOH_CONFIG_OVERRIDE='scouting/multicast/enabled=true'
ros2 run demo_nodes_cpp talker
```

```sh
# B
export ZENOH_CONFIG_OVERRIDE='scouting/multicast/enabled=true'
ros2 run demo_nodes_cpp listener
```

### Expected behavior

The `listener` on B should receive messages.

### Actual behavior

The `talker` on A complains about a missing router: `[rmw_zenoh_cpp]: Unable to connect to a Zenoh router. Have you started a router with `ros2 run rmw_zenoh_cpp rmw_zenohd`?` but then continues to `Publishing: 'Hello World: 1'` etc.

The `listener` on B does not receive messages and shows the periodic warning:
```
WARN net-0 ThreadId(02) zenoh::net::runtime::orchestrator: Unable to connect to any locator of scouted peer 8dfe45adee11bab2eedd745f7fa24506: [tcp/127.0.0.1:42649]
```

Yadunund, 2025-05-08:
Hi @christian-rauch, similar to #625 can you confirm that both hosts are using the same version of rmw_zenoh?

We're currently running Zenoh in ROS 2 nightly CI jobs with multicast enabled although all processes are within the same host: https://github.com/ros2/ros_buildfarm_config/blob/4e73515e6648eaa84909b4ff1edc2fad0d4e4b2c/rolling/ci-nightly-zenoh.yaml#L7-L8. So the issue could be either version mismatch or something on the network interface side of things. 

Regarding the warning about the router, this happens because by default rmw_zenoh sessions check for a router once before proceeding with initialization. This can be disabled by additionally setting `ZENOH_ROUTER_CHECK_ATTEMPTS: '-1'`. But this should not affect anything in your case.

christian-rauch, 2025-05-08:
`dpkg -l ros-jazzy-rmw-zenoh-cpp` shows on both PCs `0.2.3-1noble.20250429.203124`.

The network configuration is different, of course. PC A is connected via Wi-Fi and PC B is connected via Ethernet. PC B additionally has the `docker0` network interface. Not sure if this makes a difference.

Can someone reproduce this issue on a "real" setup?

closed 2025-05-08 by Yadunund

christian-rauch, 2025-05-09:
> Or simply to make all the nodes to listen on all available interfaces (including the loopback):
> 
> # A
> export ZENOH_ROUTER_CHECK_ATTEMPTS=-1
> export ZENOH_CONFIG_OVERRIDE='listen/endpoints=["tcp/0.0.0.0:0"];scouting/multicast/enabled=true'
> ros2 run demo_nodes_cpp talker
> 
> # B
> export ZENOH_ROUTER_CHECK_ATTEMPTS=-1
> export ZENOH_CONFIG_OVERRIDE='listen/endpoints=["tcp/0.0.0.0:0"];scouting/multicast/enabled=true'
> ros2 run demo_nodes_cpp listener

I guess this is related to some timeout: It takes a very long time to establish a connection.

I start the `listener` on B first:
```sh
# B (10.71.1.207/24)
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
export ZENOH_ROUTER_CHECK_ATTEMPTS=-1
export ZENOH_CONFIG_OVERRIDE='listen/endpoints=["tcp/0.0.0.0:0"];scouting/multicast/enabled=true'
. /opt/ros/jazzy/setup.bash
ros2 run demo_nodes_cpp listener
```
and then the `talker` on A:
```sh
# A (10.71.1.208/24)
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
export ZENOH_ROUTER_CHECK_ATTEMPTS=-1
export ZENOH_CONFIG_OVERRIDE='listen/endpoints=["tcp/0.0.0.0:0"];scouting/multicast/enabled=true'
. /opt/ros/jazzy/setup.bash
ros2 run demo_nodes_cpp talker
```

The listener never receives messages.

If I start the nodes the other way around, `talker` first, `listener` second, then I am losing exactly 60 messages. The first message the listener receives is: `[listener]: I heard: [Hello World: 61]`.

IMHO, there are still serious issues with the multicast functionality in `rmw_zenoh` and the new documentation does not address this issue.

JEnoch, 2025-05-13:
> At least I understand under "multicast" that messages are sent from one network host to multiple hosts at once. At least, this is how the terminology is used with other RMWs in ROS. I see, this has been updated in the documentation already.

The `scouting/multicast/enabled` config activates UDP multicast for the **scouting** protocol (i.e. endpoints discovery), not for data exchanges which will occur via the configured and discovered endpoints (TCP by default in rmw_zenoh config).

---

> IMHO, multicast on all network interfaces (listen/endpoints=["tcp/0.0.0.0:0"];scouting/multicast/enabled=true) should be the default multicast configuration for a better out-of-the-box experience for users running nodes on multiple hosts. I would naively assume that running multiple ROS nodes across multiple hosts is the most common use case for ROS.

Following the feedback of many ROS 2 users, UDP multicast has often been source of troubles and confusions in many environments not configured to support it properly. Thus the choice made by Open Robotics for this alternate RMW was precisely to avoid UDP multicast usage for a better out-of-the-box experience for newcomers.

--- 

> IMHO, there are still serious issues with the multicast functionality in rmw_zenoh and the new documentation does not address this issue.

I did few tests with the exact same commands, over my phone as a WiFi hotspot, locked in a metal box to get a 30% / -85dBm signal level. My `listener` received the `talker`  publications between 1 and 4 messages at startup. Meaning:
 - either the "Scout" UDP multicast messages are not received by either host during up-to 4 seconds
 - either the TCP connection between talker and listener takes up-to 4 seconds to establish 

Considering what you also described in #625, I think there might be something going wrong with your WiFi, network cards or OS. Hard to say without digging further in logs, or ideally in Wireshark captures on the 2 hosts.

To activate Zenoh networking logs, do:
```bash
export RUST_LOG=zenoh::net::runtime::orchestrator=trace,zenoh_link_tcp=debug,zenoh_transport=debug
```
You will see the sending and reception of Scout and Hello messages over UDP, allowing the peers to exchange their endpoints, and the TCP connection establishment. Please, let us know where you see a 60 seconds delay (UDP messages not received or TCP connection?).

commit 23015962 (2025-05-08): README: clarify ZENOH_CONFIG_OVERRIDE examples (fix #626) (#627)
commit 0b92e64c (2025-05-08): README: clarify ZENOH_CONFIG_OVERRIDE examples (fix #626) (#627) (#628)

(cherry picked from commit 23015962b0ec36cea932aab440202f368bd0ef22)
commit 8c1e636d (2025-05-08): README: clarify ZENOH_CONFIG_OVERRIDE examples (fix #626) (#627) (#629)

(cherry picked from commit 23015962b0ec36cea932aab440202f368bd0ef22)
commit cbdc225e (2025-05-08): README: clarify ZENOH_CONFIG_OVERRIDE examples (fix #626) (#627) (#630)

(cherry picked from commit 23015962b0ec36cea932aab440202f368bd0ef22)