ros2/rmw_zenoh #642: No matching queryables or not master when trying to use ROS services

g4sp3r, 2025-05-19:
Hi, first of all thank you for all your hard work.

I am trying to support ROS parameters interface for an embedded project and have hit some issues.

I am running ROS2 jazzy on raspberry-pi. rmw_zenoh is build from source (jazzy branch) with `zenoh/transport_serial` option enabled.
Embedded side is running zenoh-pico and is connected to rpi with serial link.

Following some other issues here (#125, #580, #479), pub/sub works normally between embedded and rpi, ros node is detected and services are visible:

```
ros2 node info /mcb
/mcb
  Subscribers:
    /mcb/cmd_vel: geometry_msgs/msg/Twist
  Publishers:
    /mcb/log: std_msgs/msg/String
    /mcb/odometry: nav_msgs/msg/Odometry
  Service Servers:
    /mcb/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /mcb/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /mcb/get_parameters: rcl_interfaces/srv/GetParameters
    /mcb/list_parameters: rcl_interfaces/srv/ListParameters
    /mcb/set_parameters: rcl_interfaces/srv/SetParameters
    /mcb/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
  Service Clients:

  Action Servers:

  Action Clients:
```

If I run `ros2 param list` the command locks and spins forever without any info printed to console.
Checking the router logs I see "(no matching queryables or not master)" which seems strange since the queriable is declared before with the same expresion.

```
2025-05-18T15:56:38.366368Z DEBUG ThreadId(02) zenoh::net::runtime: Zenoh Rust API v94e91794
2025-05-18T15:56:38.366382Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: 7d21e98f118f347899151c2adba90f6e
2025-05-18T15:56:38.368103Z DEBUG ThreadId(02) zenoh::net::routing::interceptor::access_control: Access control is disabled
2025-05-18T15:56:38.373851Z DEBUG ThreadId(02) zenoh::net::routing::hat::router::network: [Routers network] Add node (self) 7d21e98f118f347899151c2adba90f6e
2025-05-18T15:56:38.377139Z DEBUG ThreadId(02) zenoh::net::routing::hat::router::network: [Peers network] Add node (self) 7d21e98f118f347899151c2adba90f6e
2025-05-18T15:56:38.379784Z DEBUG ThreadId(02) zenoh::net::routing::router: New Face{0, 7d21e98f118f347899151c2adba90f6e}
2025-05-18T15:56:38.381162Z DEBUG ThreadId(02) zenoh::net::routing::dispatcher::queries: Face{0, 7d21e98f118f347899151c2adba90f6e} Declare queryable 2 (@/7d21e98f118f347899151c2adba90f6e/router/**)
2025-05-18T15:56:38.381186Z DEBUG ThreadId(02) zenoh::net::routing::dispatcher::resource: Register resource @/7d21e98f118f347899151c2adba90f6e/router/**
2025-05-18T15:56:38.384769Z DEBUG ThreadId(02) zenoh::net::routing::dispatcher::pubsub: Face{0, 7d21e98f118f347899151c2adba90f6e} Declare subscriber 3 (@/7d21e98f118f347899151c2adba90f6e/router/config/**)
2025-05-18T15:56:38.384783Z DEBUG ThreadId(02) zenoh::net::routing::dispatcher::resource: Register resource @/7d21e98f118f347899151c2adba90f6e/router/config/**
2025-05-18T15:56:38.386554Z DEBUG ThreadId(02) zenoh::net::routing::router: New Face{1, 7d21e98f118f347899151c2adba90f6e}
2025-05-18T15:56:38.387077Z DEBUG ThreadId(02) zenoh::net::runtime::orchestrator: Try to add listener: tcp/0.0.0.0:7447: ConnectionRetryConf { exit_on_failure: true, period_init_ms: 1000, period_max_ms: 4000, period_increase_factor: 2.0 }
2025-05-18T15:56:38.388319Z DEBUG ThreadId(02) zenoh::net::runtime::orchestrator: Listener added: tcp/0.0.0.0:7447
2025-05-18T15:56:38.388332Z DEBUG ThreadId(02) zenoh::net::runtime::orchestrator: Try to add listener: serial//dev/ttyAMA3#baudrate=1000000: ConnectionRetryConf { exit_on_failure: true, period_init_ms: 1000, period_max_ms: 4000, period_increase_factor: 2.0 }
2025-05-18T15:56:38.388375Z DEBUG ThreadId(02) zenoh::net::runtime::orchestrator: Listener added: serial//dev/ttyAMA3
2025-05-18T15:56:38.388574Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: serial//dev/ttyAMA3
2025-05-18T15:56:38.388582Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.1.16:7447
Started Zenoh router with id 7d21e98f118f347899151c2adba90f6e
2025-05-18T15:56:39.079364Z DEBUG acc-0 ThreadId(04) zenoh_transport::unicast::manager: Will use Universal transport!
2025-05-18T15:56:39.093418Z DEBUG acc-0 ThreadId(04) zenoh::net::routing::router: New Face{2, ae}
2025-05-18T15:56:39.093584Z DEBUG acc-0 ThreadId(04) zenoh_transport::unicast::manager: New transport opened between 7d21e98f118f347899151c2adba90f6e and ae - whatami: client, sn resolution: U32, initial sn: 4346165, qos: false, shm: None, multilink: false, lowlatency: false
2025-05-18T15:56:39.093600Z DEBUG acc-0 ThreadId(04) zenoh_transport::unicast::establishment::accept: New transport link accepted from ae to 7d21e98f118f347899151c2adba90f6e: TransportLinkUnicast { link: Link { src: serial//dev/ttyAMA3, dst: serial/4e2555fd-0400-4795-b4d0-9915718bd491, mtu: 1500, is_reliable: false, is_streamed: false }, config: TransportLinkUnicastConfig { direction: Inbound, batch: BatchConfig { mtu: 1500, is_streamed: false, is_compression: false }, priorities: None, reliability: None } }

...

2025-05-18T15:56:39.117794Z DEBUG  rx-0 ThreadId(07) zenoh::net::routing::dispatcher::resource: Register resource 0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d
2025-05-18T15:56:39.118141Z DEBUG  rx-1 ThreadId(08) zenoh::net::routing::dispatcher::queries: Face{2, ae} Declare queryable 16 (0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d)
2025-05-18T15:56:39.120930Z DEBUG  rx-0 ThreadId(07) zenoh::net::routing::dispatcher::resource: Register resource @ros2_lv/0/ae000000000000000000000000000000/0/11/SS/%/%/mcb/%mcb%list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d/::,7:,:,:,,
2025-05-18T15:56:39.121201Z DEBUG  rx-1 ThreadId(08) zenoh::net::routing::dispatcher::token: Face{2, ae} Declare token 17 (@ros2_lv/0/ae000000000000000000000000000000/0/11/SS/%/%/mcb/%mcb%list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d/::,7:,:,:,,)

...

2025-05-18T15:59:19.079014Z DEBUG  rx-0 ThreadId(07) zenoh::net::routing::dispatcher::resource: Register resource @ros2_lv/0/d295ad0dfca4ac393d307e9712b3d775/0/5/SC/%/%/_ros2cli_1507/%mcb%list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d/::,10:,:,:,,
2025-05-18T15:59:19.079068Z DEBUG  rx-0 ThreadId(07) zenoh::net::routing::dispatcher::token: Face{5, d295ad0dfca4ac393d307e9712b3d775} Declare token 16 (@ros2_lv/0/d295ad0dfca4ac393d307e9712b3d775/0/5/SC/%/%/_ros2cli_1507/%mcb%list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d/::,10:,:,:,,)

...

2025-05-18T16:19:58.226569Z DEBUG  rx-1 ThreadId(08) zenoh::net::routing::dispatcher::queries: Face{5, 67cec740b79d220628253fc4aaf1fbb9}:1 Route query for res 0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d
2025-05-18T16:19:58.226584Z TRACE  rx-1 ThreadId(08) zenoh::net::routing::hat::router::queries: compute_query_route(0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d, 0, Peer)
2025-05-18T16:19:58.226606Z DEBUG  rx-1 ThreadId(08) zenoh::net::routing::dispatcher::queries: Face{5, 67cec740b79d220628253fc4aaf1fbb9}:1 Send final reply (no matching queryables or not master)
```

I tried using zenoh-python client running on my PC connected to the same wifi as rpi. Using the same expression, the querries get routed and I can normally reach the embedded side.

Configuration is default with following changes:
 - added `serial//dev/ttyAMA3#baudrate=1000000` to `listen.endpoints` in router
 - changed `batch_size` for client and router to 2048 to match embedded side

Probably it's an error or missconfiguration on my side but I am really not sure how to debug this further.
Hopefully you can offer some guidance.

g4sp3r, 2025-05-19:
Here are also the logs from the `ros2 param list` when it tries to send the query.

```
2025-05-19T07:43:40.478740Z DEBUG ThreadId(02) zenoh::net::routing::dispatcher::resource: Register resource @ros2_lv/0/66a88905ad447c4fab355d099f26fc2f/0/9/SC/%/%/_ros2cli_2195/%mcb%set_parameters_atomically/rcl_interfaces::srv::dds_::SetParametersAtomically_/RIHS01_0e192ef259c07fc3c07a13191d27002222e65e00ccec653ca05e856f79285fcd/::,10:,:,:,,
2025-05-19T07:43:40.478758Z TRACE ThreadId(02) zenoh_transport::unicast::universal::tx: Scheduled NetworkMessage { body: Declare(Declare { interest_id: None, ext_qos: QoS { priority: Control, congestion: Block, express: false }, ext_tstamp: None, ext_nodeid: NodeIdType { node_id: 0 }, body: DeclareKeyExpr(DeclareKeyExpr { id: 18, wire_expr: WireExpr { scope: 0, suffix: "@ros2_lv/0/66a88905ad447c4fab355d099f26fc2f/0/9/SC/%/%/_ros2cli_2195/%mcb%set_parameters_atomically/rcl_interfaces::srv::dds_::SetParametersAtomically_/RIHS01_0e192ef259c07fc3c07a13191d27002222e65e00ccec653ca05e856f79285fcd/::,10:,:,:,,", mapping: Sender } }) }), reliability: Reliable } for transmission to tcp/[::1]:32966 (9bdf9b2745ceae9d46e1f2effef94d2)
2025-05-19T07:43:40.478797Z TRACE ThreadId(02) zenoh_transport::unicast::universal::tx: Scheduled NetworkMessage { body: Declare(Declare { interest_id: None, ext_qos: QoS { priority: Control, congestion: Block, express: false }, ext_tstamp: None, ext_nodeid: NodeIdType { node_id: 0 }, body: DeclareKeyExpr(DeclareKeyExpr { id: 29, wire_expr: WireExpr { scope: 0, suffix: "@ros2_lv/0/66a88905ad447c4fab355d099f26fc2f/0/9/SC/%/%/_ros2cli_2195/%mcb%set_parameters_atomically/rcl_interfaces::srv::dds_::SetParametersAtomically_/RIHS01_0e192ef259c07fc3c07a13191d27002222e65e00ccec653ca05e856f79285fcd/::,10:,:,:,,", mapping: Sender } }) }), reliability: Reliable } for transmission to tcp/127.0.0.1:7447 (816bab3ba0670161bafc4018eb0fe15f)
2025-05-19T07:43:40.478820Z TRACE ThreadId(02) zenoh_transport::unicast::universal::tx: Scheduled NetworkMessage { body: Declare(Declare { interest_id: None, ext_qos: QoS { priority: Control, congestion: Block, express: false }, ext_tstamp: None, ext_nodeid: NodeIdType { node_id: 0 }, body: DeclareToken(DeclareToken { id: 15, wire_expr: WireExpr { scope: 18, suffix: "", mapping: Sender } }) }), reliability: Reliable } for transmission to tcp/[::1]:32966 (9bdf9b2745ceae9d46e1f2effef94d2)
2025-05-19T07:43:40.478870Z TRACE ThreadId(02) zenoh_transport::unicast::universal::tx: Scheduled NetworkMessage { body: Declare(Declare { interest_id: None, ext_qos: QoS { priority: Control, congestion: Block, express: false }, ext_tstamp: None, ext_nodeid: NodeIdType { node_id: 0 }, body: DeclareToken(DeclareToken { id: 20, wire_expr: WireExpr { scope: 29, suffix: "", mapping: Sender } }) }), reliability: Reliable } for transmission to tcp/127.0.0.1:7447 (816bab3ba0670161bafc4018eb0fe15f)
2025-05-19T07:43:40.479125Z TRACE ThreadId(02) zenoh::api::session: get(0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d, AllComplete, QueryConsolidation { mode: None })
2025-05-19T07:43:40.479159Z TRACE ThreadId(02) zenoh::api::session: Register query 0 (nb_final = 2)
2025-05-19T07:43:40.479176Z DEBUG ThreadId(02) zenoh::net::routing::dispatcher::queries: Face{1, 66a88905ad447c4fab355d099f26fc2f}:0 Route query for res 0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d
2025-05-19T07:43:40.479191Z TRACE ThreadId(02) zenoh::net::routing::hat::p2p_peer::queries: compute_query_route(0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d, 0, Client)
2025-05-19T07:43:40.479220Z TRACE ThreadId(02) zenoh::net::routing::dispatcher::queries: Face{1, 66a88905ad447c4fab355d099f26fc2f}:0 Propagate query to Face{2, 816bab3ba0670161bafc4018eb0fe15f}:1
2025-05-19T07:43:40.479236Z TRACE ThreadId(02) zenoh_transport::unicast::universal::tx: Scheduled NetworkMessage { body: Request(Request { id: 1, wire_expr: WireExpr { scope: 0, suffix: "0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d", mapping: Sender }, ext_qos: QoS { priority: Data, congestion: Block, express: false }, ext_tstamp: None, ext_nodeid: NodeIdType { node_id: 0 }, ext_target: AllComplete, ext_budget: None, ext_timeout: Some(18446744073709551.615s), payload: Query(Query { consolidation: None, parameters: "", ext_sinfo: None, ext_body: Some(ValueType { ext_shm: None, encoding: Encoding { id: 0, schema: None }, payload: ZBuf { slices: [[00, 01, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]] } }), ext_attachment: Some(AttachmentType { buffer: ZBuf { slices: [[01, 00, 00, 00, 00, 00, 00, 00, 09, 4e, b3, 8a, e2, dd, 40, 18, 10, 8d, c2, 2e, 5f, a9, a4, a2, 51, af, 51, aa, d1, 62, f3, 3d, 8a]] } }), ext_unknown: [] }) }), reliability: Reliable } for transmission to tcp/127.0.0.1:7447 (816bab3ba0670161bafc4018eb0fe15f)
2025-05-19T07:43:40.479300Z TRACE ThreadId(02) zenoh::api::session: recv ResponseFinal ResponseFinal { rid: 0, ext_qos: QoS { priority: Data, congestion: Block, express: false }, ext_tstamp: None }
[... cut ...]
The only difference I can see between working and non-working query is that one is issued by Client and another by Peer:
```
found:
routing::hat::p2p_peer::queries: compute_query_route(0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d, 0, Client)

not found:
routing::hat::router::queries: compute_query_route(0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d, 0, Peer)
```

Should something be added to `DEFAULT_RMW_ZENOH_SESSION_CONFIG.json5`?

OlivierHecart, 2025-05-20:
Hello @g4sp3r !
Your problem might be due to the fact that the pico application is a Zenoh `client`. When testing with the python application, it is deployed as a `peer`. You could try to confirm that by running the python application in `client` mode.
The problems with clients might be due to `peers_failover_brokering` deactivation in rmw Zenoh router. Until we find the problem you may try to workaround the issue by reactivating `peers_failover_brokering` in the rmw Zenoh router configuration.

g4sp3r, 2025-05-20:
Hi @OlivierHecart, thanks for your time looking at this issue.

I tried to set `peers_failover_brokering=true` in the router config but the behavior is the same.

Also python test script is running in `client` mode.
```
python z-get.py  -m client  -k "0/mcb/list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d" --connect tcp/192.168.1.16:7447 
```
So to me it looks like: 
 - `client` to `client` (through the router) queries are routed OK. 
 - `peer` (ROS cli) to `client` i have the "No matching queryables " error.

I also find it strange that normal pub/sub topics are routed ok between `client` and `peer`. It would almost suggest that my configuration is good.

Maybe the liveliness expressions I am generating have some errors. But then I would expect services not being found by `ros2 node info` command.

My plan now is to try and deploy another ROS installation on host2 and try to make ROS cli nodes connect as `clients` to the rmw_zenoh router running on host1, and confirm that problem only appears with `peers`.

g4sp3r, 2025-05-20:
Another data point:

I tried running `z-get.py` both in  `peer` and `client` mode. With and without `peers_failover_brokering` the queries are routed OK to zenoh-pico `client` in all cases.

g4sp3r, 2025-05-20:
I tried running ROS cli nodes in both `peer` and `client` mode following [readme](https://github.com/ros2/rmw_zenoh/tree/rolling#connecting-to-the-zenoh-router-on-another-host) but the `"No matching queryables` error appears in both cases.

Considering that `z-get.py` works in both cases it seems that this would exclude some networking issue and the problem is more connected with RMW side of things.

Are my liveliness expressions correct?
```
@ros2_lv/0/8d1dcdedc9fac918570567fd8994ea53/0/11/SS/%/%/mcb/%mcb%list_parameters/rcl_interfaces::srv::dds_::ListParameters_/RIHS01_3e6062bfbb27bfb8730d4cef2558221f51a11646d78e7bb30a1e83afac3aad9d/::,:,:,:,,
```
Looking at the sources leaving the qos part empty (`/::,:,:,:,,`) should give default qos settings.

g4sp3r, 2025-05-20:
Ok found out the issue. It is connected with queryable completion status.  

if I update `z-get.py` to use `QueryTarget.ALL_COMPLETE` i have the same error `No matching queryables`.

I updated zenoh-pico client code with
```
 z_queryable_options_t options = {};
   options.complete = true; 
```
before declaring the queryable, and now ROS cli query gets routed to embedded side without issues.

As suspected, user error. Marking as closed.

closed 2025-05-20 by g4sp3r