ros2/rmw_zenoh #671: failure to deserialize payloads across architectures

PierreAuge, 2025-06-13:
### Operating System:

Ubuntu 24.04.02

### ROS version or commit hash:

jazzy

### RMW implementation (if applicable):

rmw_zenoh_cpp

### RMW Configuration (if applicable):

{
  mode: "router",
  connect: {
    timeout_ms: { router: -1, peer: -1, client: 0 },
    endpoints: [
      "tcp/192.168.50.173:7447"
    ],
    exit_on_failure: { router: false, peer: false, client: true },
    retry: {
      period_init_ms: 1000,
      period_max_ms: 4000,
      period_increase_factor: 2,
    },
  },
  listen: {
    timeout_ms: 0,
    endpoints: [
      "tcp/[::]:7447"
    ],
    exit_on_failure: true,
    retry: {
      period_init_ms: 1000,
      period_max_ms: 4000,
      period_increase_factor: 2,
    },
  },
  open: {
    return_conditions: {
      connect_scouted: true,
      declares: true,
    },
  },
  scouting: {
    timeout: 3000,
    delay: 500,
    multicast: {
      enabled: false,
      address: "224.0.0.224:7446",
      interface: "wlan0",
      ttl: 32,
      autoconnect: { router: ["router"], peer: ["router", "peer"], client: ["router"] },
      autoconnect_strategy: { router: { to_router: "always", to_peer: "always" } },
      listen: true,
    },
    gossip: {
      enabled: true,
      multihop: false,
      target: { router: ["router", "peer"], peer: ["router"]},
      autoconnect: { router: ["router"], peer: ["router", "peer"] },
      autoconnect_strategy: { router: { to_router: "always", to_peer: "always" } },
    },
  },

  timestamping: {
    enabled: { router: true, peer: true, client: true },
    drop_future_timestamp: false,
  },

  queries_default_timeout: 60000,

  routing: {
    router: {
      peers_failover_brokering: true
    },
    peer: {
      mode: "peer_to_peer"
    },
    interests: {
      timeout: 10000,
    },
  },
  transport: {
    unicast: {
      open_timeout: 60000,
      accept_timeout: 60000,
      accept_pending: 10000,
      max_sessions: 10000,
      max_links: 1,
      lowlatency: false,
      qos: {
        enabled: true,
      },
      compression: {
        enabled: false,
      },
    },
    multicast: {
      join_interval: 2500,
      max_sessions: 1000,
      qos: {
        enabled: false,
      },
      compression: {
        enabled: false,
      },
    },
    link: {
      tx: {
        sequence_number_resolution: "32bit",
        lease: 60000,
        keep_alive: 2,
        batch_size: 65535,
        queue: {
          size: {
            control: 2,
            real_time: 2,
            interactive_high: 2,
            interactive_low: 2,
            data_high: 2,
            data: 2,
            data_low: 2,
            background: 2,
          },
          congestion_control: {
            drop: {
              wait_before_drop: 1000,
              max_wait_before_drop_fragments: 50000,
            },
            block: {
              wait_before_close: 5000000,
            },
          },
          batching: {
            enabled: true,
            time_limit: 1,
          },
          allocation: {
            mode: "lazy",
          },
        },
      },
      rx: {
        buffer_size: 65535,
        max_message_size: 1073741824,
      },
      tls: {
        root_ca_certificate: null,
        listen_private_key: null,
        listen_certificate: null,
        enable_mtls: false,
        connect_private_key: null,
        connect_certificate: null,
        verify_name_on_connect: true,
        close_link_on_expiration: false,
      },
    },
    shared_memory: {
      enabled: false,
      mode: "lazy",
    },
    auth: {
      usrpwd: {
        user: null,
        password: null,
        dictionary_file: null,
      },
      pubkey: {
        public_key_pem: null,
        private_key_pem: null,
        public_key_file: null,
        private_key_file: null,
        key_size: null,
        known_keys_file: null,
      },
    },
  },

  adminspace: {
    enabled: true,
    permissions: {
      read: true,
      write: false,
    },
  },

}

### Client library (if applicable):

ROS2 commandline

### 'ros2 doctor --report' output

ros2 topic echo /test

2025-06-13T02:26:06.359843Z ERROR rx-1 ThreadId(08) zenohc::serialization: Failed to deserialize the payload: deserialization error
thread 'rx-1' panicked at library/core/src/panicking.rs:136:5:
panic in a function that cannot unwind
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread caused non-unwinding panic. aborting.
Aborted (core dumped)

### Steps to reproduce issue

On machine 1 with x86 architecture publish a topic of any kind example: ros2 topic pub /test std_msgs/msg/String "data: test"

On machine 2 with arm64 architecture subscribe to or echo the topic and see deserialization error. example: ros2 topic echo /test

### Expected behavior

The expected behavior should to properly subcribe to and deserialize the message.

### Actual behavior

panic and mayhem

`2025-06-13T02:26:06.359843Z ERROR rx-1 ThreadId(08) zenohc::serialization: Failed to deserialize the payload: deserialization error
thread 'rx-1' panicked at library/core/src/panicking.rs:136:5:
panic in a function that cannot unwind
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread caused non-unwinding panic. aborting.
Aborted (core dumped)`

### Additional information

WTF? ;)

Yadunund, 2025-06-13:
What is the version of rmw_zenoh on the x86 and arm64 machines respectively? Was rmw_zenoh built from source or are you using binaries?

PierreAuge, 2025-06-13:
ARM64 binary from apt - ros-jazzy-rmw-zenoh-cpp 0.2.3-1noble.20250430.003608
X86 binary from apt - ros-jazzy-rmw-zenoh-cpp 0.2.3-1noble.20250429.203124

Apologies I should have included that right off the bat.

PierreAuge, 2025-06-14:
After switching to the current master branch in the repo and compiling on both arm and x86 I was able to get it working.

Yadunund, 2025-06-15:
> ARM64 binary from apt - ros-jazzy-rmw-zenoh-cpp 0.2.3-1noble.20250430.003608
X86 binary from apt - ros-jazzy-rmw-zenoh-cpp 0.2.3-1noble.20250429.203124

Interesting. I was expecting to see different versions and point you to https://github.com/ros2/rmw_zenoh/issues/637. tldr there is a wire breaking change between 0.2.4 and 0.2.3 but since both your platforms are on the same version i'm not sure what could have gone wrong....

We're going to release the latest jazzy versions soon so  the equivalent the versions should match whats on `main` for the most part.

PierreAuge, 2025-06-19:
Apologies I dropped out there. I've been on the road for work so I haven't had time to rehash this.

At the time when I RUST_BACKTRACE=1 I got no further output.

JEnoch, 2025-06-30:
Hi @PierreAuge ,
[Jazzy was synced yesterday](https://discourse.ros.org/t/new-packages-for-jazzy-jalisco-2024-07-29/44516) and includes rmw_zenoh 0.2.5. 
Could you please update your 2 machines with this version and let us know is you still see the deserialization error ?

PierreAuge, 2025-06-30:
I'm currently in Germany I'll give it a whirl when I'm back in the lab next week. Maybe I'll remote in if I get a chance. But I'll let you know for sure.

peci1, 2025-08-08:
I see the very same behavior even between two amd64 machines on 0.2.5 (Jazzy).

JEnoch, 2025-08-12:
@peci1 thank you for your confirmation.

Before closing this ticker, I would like to know:
@PierreAuge did you manage to test with `0.2.5` on your 2 hosts ?

JEnoch, 2025-08-13:
> It's now working without issue thanks.

Thanks for your confirmation.
I'm closing this issue, then.

closed 2025-08-13 by JEnoch