ros2/rmw_zenoh #674: [Question] Clarification regarding comment about copying default config

oysstu, 2025-06-18:
https://github.com/ros2/rmw_zenoh/blob/fce38ddb99358dad479a184442292682c8590423/rmw_zenoh_cpp/config/DEFAULT_RMW_ZENOH_SESSION_CONFIG.json5#L3

Would it be possible to clarify the above comment. Are these not the default values used by rmw_zenoh_cpp? What would be considered good practice? 

I've copied the default config, overridden a few fields, and removed all the entries I have not changed. Is this what the comment is hinting at?

YuanYuYuan, 2025-06-19:
Hi @oysstu! This comment is just copied from the upstream. https://github.com/eclipse-zenoh/zenoh/blob/6f9cfa0581f110d6e043e984f5b08300851a5eb6/DEFAULT_CONFIG.json5#L3. It's not appropriate to state like this in rmw_zenoh since there's no _config.rs_ in rmw_zenoh.

Regarding your question, you can remove the disabled items while preserving the others (and your modifications). Removing the other fields, even if you don't modify them, would change the suggested behavior for rmw_zenoh session and router.

oysstu, 2025-06-19:
I see, thanks for clearing that up :+1: