ros2/rmw_zenoh #686: Error deserializing message in ROS 2 Service when querying from Zenoh-Pico application

VintageDoggo, 2025-06-25:
### Operating System:

```Linux 6.6.23 (Yocto generated)```

### ROS version or commit hash:

Jazzy

### RMW implementation (if applicable):

rmw_zenoh_cpp

### RMW Configuration (if applicable):

Default configuration for Zenoh router and sessions. 

### Client library (if applicable):

rclcpp, cyclonedds-cxx

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
 NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,RUNNING,LOOPBACK>
mtu          : 65536
inet         : 10.88.0.2
inet4        : ['10.88.0.2']
ether        : 42:1a:21:3b:29:3e
inet6        : ['fe80::401a:21ff:fe3b:293e%eth0']
netmask      : 255.255.0.0
device       : eth0
flags        : 4163<UP,BROADCAST,MULTICAST,RUNNING>
mtu          : 1500
broadcast    : 10.88.255.255

   PACKAGE VERSIONS
log_srv                                   : latest=N/A, local=0.0.0
custom_interfaces                         : latest=N/A, local=0.0.0
diagnostic_msgs                           : latest=5.3.6, local=5.3.6
ament_cmake_xmllint                       : latest=0.17.2, local=0.17.2
ament_cmake_target_dependencies           : latest=2.5.4, local=2.5.4
ament_cmake_version                       : latest=2.5.4, local=2.5.4
ament_xmllint                             : latest=0.17.2, local=0.17.2
console_bridge_vendor                     : latest=1.7.1, local=1.7.1
rosidl_pycommon                           : latest=4.6.5, local=4.6.5
rcl_logging_spdlog                        : latest=3.1.1, local=3.1.1
ros_environment                           : latest=4.2.1, local=4.2.1
rclcpp_components                         : latest=28.1.10, local=28.1.9
ament_flake8                              : latest=0.17.2, local=0.17.2
ros2action                                : latest=0.32.4, local=0.32.4
ament_cmake_export_interfaces             : latest=2.5.4, local=2.5.4
rosidl_typesupport_c                      : latest=3.2.2, local=3.2.2
ament_index_python                        : latest=1.8.1, local=1.8.1
action_msgs                               : latest=2.0.2, local=2.0.2
actionlib_msgs                            : latest=5.3.6, local=5.3.6
ament_uncrustify                          : latest=0.17.2, local=0.17.2
rclcpp_action                             : latest=28.1.10, local=28.1.9
rosidl_adapter                            : latest=4.6.5, local=4.6.5
ros_core                                  : latest=0.11.0, local=0.11.0
ament_cmake_cpplint                       : latest=0.17.2, local=0.17.2
ament_lint_common                         : latest=0.17.2, local=0.17.2
sros2                                     : latest=0.13.3, local=0.13.3
[... 247 lines cut ...]
module custom_interfaces {
  module srv {
    struct LogServiceInput_Request {
      string input;
    };
    struct LogServiceInput_Response {
      string output;
    };
  };
};
```

The query does nothing with the input Request, it simply returns a value. For debugging purposes, I added the string `abc`. 
With this, my buffer holds the following values:
```
[ 0x00 0x01 0x00 0x00 // CDR headers and options -- little endian
0x04 0x00 0x00 0x00 // payload length
0x61 0x62 0x63 0x00 // payload 'abc'
0x00 0x00 0x00 0x00 0x00 // The rest of my buffer is initialized to zero, since the size of the buffer is taken from the `sizeof()` of my `srv` interface. 
0x00 0x00 0x00 0x00 0x00
0x00 0x00 0x00 0x00 ]
```
I have also tried readjusting the size so that only 12 octets are sent. 

I am aware of the wire break changes between `rmw_zenoh` and `zenoh-pico`, but I suspect this is not that, as I also tried with `v1.3.0` and `v1.3.4`. Is my payload malformed or is there an issue between FastCDR and CycloneDDS's interface representation that is causing the deserialization issue?

VintageDoggo, 2025-06-28:
After bumping to Kilted, everything works as expected.

closed 2025-06-28 by VintageDoggo

Yadunund, 2025-06-30:
@VintageDoggo glad to hear the issue is resolved.