ros2/rmw_zenoh #689: Using transport_serial by default for ROS packages

g4sp3r, 2025-06-28:
### Description

Hi,

We are trying to use rmw_zenoh for our ROS2 based systems. Embedded hardware is running zenoh-pico and is connected to the host with serial link. 
Currently we need to compile the ROS package by hand with `zenoh/transport_serial` compile option set.

Are there any plans to have serial support enabled by default for released ROS packages? 

### Motivation

...

### Design / Implementation Considerations

...

Yadunund, 2025-06-30:
@g4sp3r thanks for the ticket. We've had requests enable other link types by default, see https://github.com/ros2/rmw_zenoh/pull/660, and i've left a comment on a couple ways we can approach it. https://github.com/ros2/rmw_zenoh/pull/660#issuecomment-2950627720. 

Feel free to chime in there and even better if you can help benchmark the increase in build time and memory footprint from enabling all features.

g4sp3r, 2025-06-30:
@Yadunund thanks for the pointing me in the right direction. I missed that thread.

PR #692 (2025-07-10): Change zenoh-c features to use its default + shared-memory + transport_serial
## Description

`rmw_zenoh` is currently building `zenoh-c` and `zenoh` with a limited subset of features:
- `shared-memory`
- `zenoh/transport_compression` 
- `zenoh/transport_tcp` 
- `zenoh/transport_udp` 
- `zenoh/transport_tls`

This PR changes the set of features to include:
- `zenoh-c` [default features](https://github.com/eclipse-zenoh/zenoh-c/blob/1.4.0/Cargo.toml#L49C1-L60C2) currently including:
  - `auth_pubkey`
  - `auth_usrpwd`
  - `transport_multilink`
  - `transport_compression`
  - `transport_quic`
   [... cut]
closed 2025-07-10 by Yadunund