ros2/rmw_zenoh #690: multicast discovery is instable to find endpoints, that fails to some ros2cli tests

fujitatomoya, 2025-06-28:
### Operating System:

```5.15.0-139-generic #149~20.04.1-Ubuntu SMP Wed Apr 16 08:29:56 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

rolling source build

### RMW implementation (if applicable):

rmw_zenoh_cpp

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

see the following CI failures.
the reason of failures are that command line test operation does not find the expected endpoints that is supposed to be online.

- https://ci.ros2.org/job/ci_linux/24134/testReport/junit/ros2action.ros2action.test/test_echo/test_echo/
- https://ci.ros2.org/job/ci_linux-aarch64/18207/testReport/junit/ros2param.ros2param.test/test_verb_load/test_verb_load/

### Expected behavior

those tests should pass.

### Actual behavior

those tests fail sometimes.

fujitatomoya, 2025-07-01:
https://github.com/ros2/ros2cli/pull/1062 should address this issue.

Yadunund, 2025-07-31:
Still seeing test [failures](https://build.ros2.org/job/Rci__nightly-zenoh_ubuntu_noble_amd64/133/testReport/ros2topic.ros2topic.test/test_echo_pub/test_echo_pub/) even after https://github.com/ros2/ros2cli/pull/1062 was merged. 🤔

cottsay, 2025-07-31:
It looks like that test (test_echo_pub) didn't get isolation because unlike the other ros2cli tests which are run on a per-RMW basis, that one is run using the typical runtime-configured RMW. I'll open a PR to add isolation.

Yadunund, 2025-07-31:
Thanks @cottsay.

Yadunund, 2026-11-14:
Closing as we've adopted isolated tests throughout that rely on the router for discovery.

closed 2026-11-14 by Yadunund