ros2/rmw_zenoh #693: topic name /parameter_events not found in topic_map. Report this.

siddux, 2025-07-02:
I've found that when I run several nodes, sometimes I get a bunch of messages like this (not sure if all of them are related to the same issue):

```
topic name /parameter_events not found in topic_map. Report this.
topic name /controller_manager/list_controllers not found in topic_map. Report this.
qos :2:,10:,:,:,, not found in for topic type controller_manager_msgs::srv::dds_::ListControllers_. Report this.
Query queue depth of 10 reached, discarding oldest Query for client for 200/peripherals/environmental_sensors/get_data/mk4_interfaces::srv::dds_::GetEnvironmentData_/TypeHashNotSupported
```

And this causes some my services and topics to stop communicating between them.

I've seen this is similar to #408 and #476 although in my case it does not happen when I launch it but later on. Is there any way to prevent it or we need to wait for the next release?

siddux, 2025-07-03:
So I've been digging a little bit more into the issue (still on the 0.1.1 version as changing that it will require some time) and the issue is the following:

I have my robot running the zenoh router and I have a laptop connected over wifi running a session (using the peer mode). On normal operation everything works really well and I keep the load of the CPU to quite low values between 4 and 6 (for reference, the same setup with cyclone DDS keeps it around 10-12).

The problem appears if I restart the graphical application on the laptop, where the load on the robot CPU rises immediately to 20-30 and the comms, even with internal nodes, becomes extremely laggy and nothing works until I reboot it. In comparison, with cyclone I don't have this massive increment, just a minor spike that quickly goes back to normal values.

Note that when I restart the application, the session always keeps running as it runs on the background and doesn't depend on the app. I'm not sure if there's any parameter on the config of either the router or the peer that may help.

Yadunund, 2025-07-29:
@siddux we released `0.1.4` of `rmw_zenoh` on Humble. Please update your binaries and try again. If the issue persists, feel free to reopen the ticket.