ros2/rmw_zenoh #731: Session vs Route did not match any configured ACL subject

Thongtdextra, 2025-08-07:
### Issue Type

- [x] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Humble

### Description

Description of zenoh operation:
At host A:
1 Session has ID = SessionA responsible for communication between internal device and RouterZenoh of hostA.
1 Route Zenoh has ID = RouteA responsible for communication between device and RouterZenoh B with ID = RouteB.

At HostB:
1 Session has ID = SessionB responsible for communication between internal device and RouterZenoh of hostingB.
1 Route Zenoh has ID = RouteB responsible for communication between device and RouterZenoh B with ID = RouteA.

I want to use access control to configure a whitelist of topics that are allowed to communicate between RouteA and RouteB over the Wg0 interface.

Here is the Route and Session configuration at A and B. And i still meet errorr: Session vs Route did not match any configured ACL subject. (in my config, HostA aka client will set access control, and other config have default access control (allow all traffic).
i also check https://zenoh.io/docs/manual/access-control/ but my nothing change.

### Screenshots or Examples (if applicable)

https://discordapp.com/channels/914168414178779197/1396422652050145290
here is my thread in zenoh discord

JEnoch, 2025-08-19:
Here are the conclusions of the discussion that occurred on Zenoh Discord in https://discordapp.com/channels/914168414178779197/1396422652050145290 :

Such a log:
`2025-08-07T02:33:20.451253Z  INFO acc-0 ThreadId(04) zenoh::net::routing::interceptor::access_control: 2bfcfe5d2c894a4946385f63f0430210 did not match any configured ACL subject. Default permission 'Allow' will be applied on all messages`
is not an error indicating that no access control is applied at all.
When logged by the router, it refers to the link between the Node and the Router over the loopback. As there is no ACL configured for this link, as the Node s allowed to communicate with the Router.
If the ACL is configured over the WiFi interface, it will apply here.

The reason ACL configuration were not working was because of the introduction of Zenoh AdvancedPublisher to support `TRANSIENT_LOCAL` and `RELIABLE`. It adds a Queryable on such a key expression:
`0/chatter/std_msgs::msg::dds_::String_/TypeHashNotSupported/@adv/sub/18bf4236bf6b467ee3ec28d43d182289/15/_`
This Queryable allows an AdvancedSubscriber to retrieve missed publications.

And that's what happens: the publication is well blocked by the Router's ACL, but the subscriber detects the missing message and queries it from the AdvancedPublisher. The reply to this query is not blocked by the Router's ACL and the data goes to the subscriber.

To fix this issue the ACL must be configured to also denies the Queryable declaration and its replies the `"/@adv/"` key expression, as such:
```json5
access_control: {
  enabled: true,
  default_permission: "allow",   // allow all by default
  /// List of rules
  rules: [
    {
      // Rule identifier
      id: "my_rule",
      // Deny publications ("put" Zenoh operation) and missed data retrieval in egress on WiFi interface for a Zenoh key expression
      permission: "deny",
      messages: [ "put", "declare_queryable", "reply" ],
      flows:["egress"],
      key_exprs: [
        // The Zenoh key expression used for the "/chatter" topic
        //"0/chatter/std_msgs::msg::dds_::String_/RIHS01_df668c740482bbd48fb39d76a70dfd4bd59db1288021743503259e948f6b1a18"
        "*/chatter/**",  
        // The Zenoh key expression used for retrieval of missed data on the "/chatter" topic
        //"0/chatter/std_msgs::msg::dds_::String_/RIHS01_df668c740482bbd48fb39d76a70dfd4bd59db1288021743503259e948f6b1a18/@adv/pub/*/*/*"
        "*/chatter/**/@adv/pub/**"
      ],
    },
  ],
  /// list of subjects:
  subjects: [
    {
      // Interface identifier
      id: "my_interface",
      interfaces: ["eno1"],
    },
  ],
  /// apply rule to subject in the policies list
  policies: [
    {
      rules: ["my_rule"],
      subjects: ["my_interface"],
    },
  ],
},
```

In addition, to also deny the topic to be discovered from a remote host, the `"liveliness_token"` messages on such key expressions could be denied:
- for a topic without namespace: `"@ros2_lv/**/%chatter/**"`
- for a topic with namespace: `"@ros2_lv/**/$*%chatter/**"`
- for a set of topics with `mavros` as a namespace: `"@ros2_lv/**/%mavros%$*/**"`

closed 2025-08-19 by JEnoch

Thongtdextra, 2025-09-05:
Continuing with access control, the issue of topic ACL has been explained. However, when I want to deny the node, I encounter a problem.
At host A, B, C, I run a node named plan_upload_node. The node at HostA publishes the topic "hostA". The node at HostB publishes the topic "hostB",...
I want to deny the node plan_upload_node and not let this node communicate to the server to avoid duplicate node names.
I tried access control at the router config file:
key_exprs: [
"@ros2_lv/**/plan_upload_node/**"
],
Here is the log when running the node plan_upload_node: 

[log.txt](https://github.com/user-attachments/files/22168760/log.txt)