ros2/rmw_zenoh #744: Advanced Cache interaction with zenoh-pico

VintageDoggo, 2025-08-13:
Hello,

I'm currently developing a zenoh-pico application that needs to retrieve historical data from a ROS node. I have set my node's durability to transient local and to be reliable so that an `AdvancedPublisher` is constructed. 

From my understanding, the AdvancedPublisher starts an underlying Zenoh queryable on the same Key Expression as the publisher with an added suffix "/@adv/pub/zid/eid/_". 

From my zenoh-pico application I send a query with the attachment and options structure discussed in the design page, however I get no responses. When checking the packets between 2 ROS nodes with an AdvancedPublisher and an AdvancedSubscriber respectively, I can see that the query is done with the selectors `_max` and `_anyke`, and so I tried implementing this in my pico application. This resulted in fetching only the latest value from the cache.

I believe this is more likely due to my implementation, but it is still unclear to me why the `_anyke` selector has to be used. 
Any comments or guidance are hugely appreciated! 

System information

* zenoh-pico: v1.5.0
* rmw_zenoh: 0.8.1 - Kilted

YuanYuYuan, 2025-08-14:
Hi @VintageDoggo! To fetch the historical data in a ROS way, you should use an advanced subscriber with the corresponding [options](https://github.com/ros2/rmw_zenoh/blob/47b724b688a6b59bcc51b7b4089c470ee542f279/rmw_zenoh_cpp/src/detail/rmw_subscription_data.cpp#L190-L198) rather than a querier.

JEnoch, 2025-08-14:
Hi @VintageDoggo,
the Advanced Pub/Sub for zenoh-pico are currently a work in progress: https://github.com/eclipse-zenoh/zenoh-pico/pull/1011 
You may want to wait for their completion, or you could check out the code and potentially review or contribute to the effort!

I’ll close this ticket since the question is more relevant to zenoh-pico than to rmw_zenoh.

closed 2025-08-14 by JEnoch

VintageDoggo, 2025-08-14:
Hi @YuanYuYuan! Thank you! Since my other system is not running ROS I thought I could tap into the queryeable. But I'll look into the Advanced Subscriber.

VintageDoggo, 2025-08-14:
Hi @JEnoch! Great, I've been keeping my eye on the PR, but I'll try to contribute to it!

On a more rmw_zenoh note, I'm still confused about the selectors used in the AdvancedPublisher. Does it use the same as the base Rust implementation (`_age`, `_max`, and `_sn)?

JEnoch, 2025-08-14:
Yes, `rmw_zenoh` uses `zenoh-c` which is a binding over `zenoh` Rust. 
Thus `rmw_zenoh` uses the Advanced Pub/Sub Rust implementation underneath.

In the [AdvancedSubscriber](https://github.com/eclipse-zenoh/zenoh/blob/1.4.0/zenoh-ext/src/advanced_subscriber.rs):
- the `_time` parameter is added when calling `params.set_time_range(...)`
- the `_max` parameter is added with `params.insert("_max", ...)`
- the `_sn` parameter is added in the `seq_num_range(start: Option<u32>, end: Option<u32>)` function 
- the `_anyke` parameter is added when calling [`accept_replies(ReplyKeyExpr::Any)`](https://docs.rs/zenoh/latest/zenoh/session/struct.SessionGetBuilder.html#method.accept_replies)