ros2/rmw_zenoh #757: Shared-memory transport not reducing loopback traffic

faseelmo, 2025-08-25:
I tested the shared-memory transport option in rmw_zenoh, but I don’t see any reduction in loopback traffic when it is enabled.

#### Steps to reproduce: 
1. Set the environment variable
```export ZENOH_CONFIG_OVERRIDE='transport/shared_memory/enabled=true'```   
2. Start the zenoh router.
3. Launch the Nav2 stack.
4. Monitor loopback traffic with Wireshark.

#### Observed behavior:
- Average loopback traffic is around 6 MB/sec, regardless of whether shared memory is enabled or not.
- I tested this both with binary installation and by building from source and the results are the same.

#### Expected behavior:
- With shared-memory transport enabled, I expected the loopback traffic to be significantly reduced. 

#### Comparison with FastDDS:
Running the same Nav2 setup with FastDDS (which I believe uses shared memory by default), the loopback traffic is only about 0.15 MB/sec.

#### Environment 
ROS 2: Jazzy 
rmw_zenoh: 7338b92

Is there something I’m missing in enabling shared memory in rmw_zenoh? Or could this be a bug where the shared-memory transport isn’t being used as expected?

faseelmo, 2025-08-25:
I didn't see any difference after changing the the message size threshold. 

To create a more reproducible version of my issue, I tested this using the Nav2 tutorial from [the official docs](https://docs.nav2.org/getting_started/index.html#installation)

I ran three different tests:

**FastDDS**
Average bandwidth: ~400,000 Bytes/sec (loopback).

**Zenoh (shared memory disabled)**
Average bandwidth: ~800,000 Bytes/sec (loopback).

**Zenoh (shared memory enabled, message size threshold = 64)**
Average bandwidth: ~800,000 Bytes/sec (loopback).

There was no difference between Zenoh with or without shared memory enabled.

Now I’m wondering if I may not be building from source correctly.
I know that I need to add the feature in the Cargo flags, but shared memory already seems to be enabled [here](https://github.com/ros2/rmw_zenoh/blob/7338b92d40f44903c920326a71bc3041aa0c5738/zenoh_cpp_vendor/CMakeLists.txt#L18)

JEnoch, 2025-08-26:
> Now I’m wondering if I may not be building from source correctly.
 
To check is SHM is enabled, check if you have some `/dev/shm/*.zenoh` files while running your nodes.
Note that 1 file could remain after a run, but it would be cleaned up as soon as you start again the router or any other node.

> I know that I need to add the feature in the Cargo flags, but shared memory already seems to be enabled [here](https://github.com/ros2/rmw_zenoh/blob/7338b92d40f44903c920326a71bc3041aa0c5738/zenoh_cpp_vendor/CMakeLists.txt#L18)

Right, the feature is enabled by default. No need to change the `zenoh_cpp_vendor/CMakeLists.txt`.

---

I did run the Nav2 tb3_simulation in a Docker and analyzed Wireshark capture over the loopback. I was using Jazzy and building `rmw_zenoh` commit 7338b92 from sources (no feature change).

First thing to notice is that the capture over loopback also include the Gazebo traffic over TCP (non-zenoh).
If for FastDDS use case you filter on UDP, you won't count this traffic. While for Zenoh, if you filter on TCP, you will count this traffic.
In my captures, I estimates Gazebo traffic at ~500,000 Bytes/sec.

Here are my captures filtering only the Zenoh traffic, with 3 different configurations:

<details>
  <summary><b>Shared memory disabled => ~400,000 Bytes/sec</b></summary>

[zenoh-noshm.pcap.zip](https://github.com/user-attachments/files/21988132/zenoh-noshm.pcap.zip)

[image]
</details>

<details>
  <summary><b>Shared memory enabled, message size threshold = 512 => ~300,000 Bytes/sec</b></summary>

[zenoh-shm-512.pcap.zip](https://github.com/user-attachments/files/21988152/zenoh-shm-512.pcap.zip)

[image]
</details>

<details>
  <summary><b>Shared memory enabled, message size threshold = 16 => ~350,000 Bytes/sec</b></summary>

[zenoh-shm-16.pcap.zip](https://github.com/user-attachments/files/21988159/zenoh-shm-16.pcap.zip)

[image]
</details>

Comparing in details with the [zenoh-dissector](https://github.com/eclipse-zenoh/zenoh-dissector)*:
- the big 49238 bytes frames (which are Zenoh fragments of data) diseapear when shared memory is enabled. However, there are only a few of those frames, and seem to correspond to a publication every 2 seconds.
- with message size threshold = 16, the only `Push` messages I see have a 12 bytes payload. Still, there are a lot and cause this ~350,000 Bytes/sec traffic.

So I believe I can conclude that shared-memory is working, however with limited impact in the case of Nav2 tb3_simulation because the bulk of the traffic is Gazebo and 12 bytes data.

Note that lowering the `ZENOH_SHM_MESSAGE_SIZE_THRESHOLD` can be counter-productive for the latency of "small" messages, as the management of SHM (allocation, mounting, garbage-collection...) adds an overhead which can become greater than the time to send a small payload over the loopback.
With current version (zenoh 1.5.0) on typical x86_64 laptops we measured that below 512 bytes of payload (CDR-serialized) shared-memory doesn't bring a latency gain wrt. loopback. The results will indeed depends the hardware.
We plan to make improvements in SHM efficiency that will lower this threshold.

---

***Note for zenoh-dissector usage**:
By default Wireshark will decode only the traffic on TCP port 7447. (i.e. with the router). As ROS nodes are communicating peer-to-peer, they use different ports.
To make Wireshark decode the messages in my already-filtered captures:
 - right-click on any `TCP` message 
 - Decode as...`
 - In `Current` column, replace `(none)` with `Zenoh`.
 - `OK`
This makes all the messages using the same TCP port number to me decoded as Zenoh. Repeat until all frames are marked as `Zenoh`.

faseelmo, 2025-08-27:
Thanks a lot for the detailed reply. A lot of the networking details make sense to me now.
I managed to fix the issue by inspecting the frames in Wireshark. Initially, I had only exported
`export ZENOH_CONFIG_OVERRIDE='transport/shared_memory/enabled=true'` in the terminal where I was running the router. I didn’t realize that I also needed to do this in the terminal where I run the nodes as well. After doing that, I no longer saw the larger frames appear in Wireshark.

closed 2025-08-27 by faseelmo

JEnoch, 2025-08-27:
Thanks for your feedback! It shows that we need to clarify this point in the documentation.
Done in #759.