ros2/rmw_zenoh #763: Large CPU spike in rmw_zenohd when a node restarts on another host

oysstu, 2025-08-28:
### Operating System:

Ubuntu 24.04

### ROS version or commit hash:

Kilted

### RMW Configuration (if applicable):

Two hosts with one zenohd router running on each with DEFAULT_RMW_ZENOH_ROUTER_CONFIG.json5 except an connect: endpoints entry between the routers.

### Steps to reproduce issue

Restarting a node (in this case an ublox driver) on host B causes a significant spike in CPU utilization of host A. The ublox node sets up a few topics of course, but I'm surprised that another host can be this severely affected by a node restarting on another host. See the following btop output. Every time the node on host B restarts, all cores on this CPU are loaded fully for about a second. It's the rmw_zenohd process that causes the CPU spike.

[image]

Thoughts? Are there any configuration options that may help alleviate this?

### Expected behavior

Host B should not affect host A by simply restarting a node. 

### Actual behavior

100% CPU usage across four cores for about a second. This severely impacts other running processes.

oysstu, 2025-08-29:
There's also a spike in CPU usage when restarting the same process on localhost. It's smaller in magnitude though, but I'm not sure if that is due to it sharing CPU utilization with the process restarting or if rmw_zenohd actually uses less CPU.

YuanYuYuan, 2026-01-08:
> In my case, the issue stopped occurring after I re-applied the customizations I had made to the configuration file based on the latest version in this repository.

Thanks @Tacha-S !

Closes this as resolved.