ros2/rmw_zenoh #769: Possible memory leak when utilizing shm feature on humble

0b10010010, 2025-09-03:
### Operating System:

```6.8.0-60-generic #63~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 22 19:00:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux```

### ROS version or commit hash:

`humble`

### RMW implementation (if applicable):

`rmw_zenoh_cpp`

### RMW Configuration (if applicable):

shm enabled for the router and each session config file. Set the allocation size to 1GB with `ZENOH_SHM_ALLOC_SIZE=1073741824`.

### Client library (if applicable):

`rclcpp`

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

1. Configure docker compose to use shared memory with sufficient `shm_size`. Then publish and subscribe large data across two containers.
2. Using `docker stats` to monitor resource usage
3. Disable `shm` feature and repeat step 2.

### Expected behavior

The memory usage should be constant and should not grow over time.

### Actual behavior

When inspecting with `docker stats`, memory usage grows over time with the shared memory feature enabled.

YuanYuYuan, 2025-09-04:
It seems not able to be reproduced on the rolling branch.

0b10010010, 2025-09-04:
Hi @YuanYuYuan,
thanks for checking. I stumbled upon [this](https://github.com/ros2/rmw_zenoh/pull/142) which has been merged to `rolling`. But I can't find any reference to it being backported. Is it safe to assume that I am seeing something similar?

YuanYuYuan, 2025-09-05:
Thanks @0b10010010! I will look into this 👀

closed 2025-09-08 by 0b10010010

YuanYuYuan, 2025-09-08:
Hi @0b10010010, never mind. Thank you for the update!

0b10010010, 2025-09-08:
Hi @JEnoch,

thanks for providing the insight. As I have very limited exposure to the inner workings of `rclcpp`, the information I provided assumes a lot of things and is based on pure observation.

The statement I made should've mentioned it "seems" like so. Because I am unsure what else could be causing the discrepancy in the total memory used. When the only variable is the number of threads used.

I would much appreciate if there is an obvious explanation. Please let me know when you get a chance.