ros2/rmw_zenoh #783: Service call reports z_reply_is_ok returned False Reason: Timeout

mini-1235, 2025-09-09:
### Generated by Generative AI

No

### Operating System:

Linux XPS-15-9520 6.8.0-78-generic #78-Ubuntu SMP PREEMPT_DYNAMIC Tue Aug 12 11:34:18 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux

### ROS version or commit hash:

Rolling binary http://packages.ros.org/ros2/ubuntu noble/main amd64 ros-rolling-rmw-zenoh-cpp amd64 0.9.0-1noble.20250822.215306

### RMW implementation (if applicable):

zenoh

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

1.  run nav2 with 
```
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=0
````

2. wait until initial transform timeout and see the error from zenoh

### Expected behavior

Receive a response of false from the change_state service server

### Actual behavior

The change_state service does not return false indicating state is unable to change, instead I see a couple of errors from the zenoh side
```
[component_container_isolated-8] [ERROR 1757390757.072415008] [rmw_zenoh_cpp]: z_reply_is_ok returned False Reason: Timeout (add_new_query() at ./src/detail/rmw_service_data.cpp:252)
[component_container_isolated-8] 2025-09-09T04:05:58.074188Z  WARN ThreadId(12) zenoh::api::session: Received ReplyData for unknown Query: 77
[component_container_isolated-8] 2025-09-09T04:05:58.074224Z  WARN ThreadId(12) zenoh::api::session: Received ResponseFinal for unknown Request: 77
[component_container_isolated-8] [ERROR 1757390758.431290576] [rmw_zenoh_cpp]: Query queue depth of 10 reached, discarding oldest Query for service for 0/lifecycle_manager_navigation/is_active/std_srvs::srv::dds_::Trigger_/RIHS01_eeff2cd6fa5ad9d27cdf4dec64818317839b62f212a91e6b5304b634b2062c5f (add_new_query() at ./src/detail/rmw_service_data.cpp:252)
```

and cannot get any result from service server

### Additional information

I was trying to setup rmw_zenoh for Nav2 in this PR https://github.com/ros-navigation/navigation2/pull/5516, one of the system tests is failing here https://app.circleci.com/pipelines/github/ros-navigation/navigation2/16036/workflows/7e2c6499-4b74-4e47-9c88-cf5953de2fd1/jobs/46834. Note that this test passes on fastrtps / cyclonedds, and only fails on zenoh.

I will illustrate the working flow here:
1. After we launch the tb3 simulation, we are changing the states for the lifecycle nodes, the planner server, when calling `on_activate`,  https://github.com/ros-navigation/navigation2/blob/ab4aaf863aea6a7d1e67e0c2ad536ec1f0c8ccf7/nav2_planner/src/planner_server.cpp#L180 will activate global costmap. However, for global costmap to be activated we typically need a transform from map -> odom -> robot_base_frame, which is unavailable unless we provide an initial pose.  So, planner server should fail to change state until the initial pose is passed.
2. Suppose we never provide the initial pose, then costmap ros should fail to activate when it reaches initial transform timeout, https://github.com/ros-navigation/navigation2/blob/ab4aaf863aea6a7d1e67e0c2ad536ec1f0c8ccf7/nav2_costmap_2d/src/costmap_2d_ros.cpp#L303-L310, planner will also return failure here https://github.com/ros-navigation/navigation2/blob/ab4aaf863aea6a7d1e67e0c2ad536ec1f0c8ccf7/nav2_planner/src/planner_server.cpp#L181-L183
3. The lifecycle manager, who was trying to change planner server's state earlier https://github.com/ros-navigation/navigation2/blob/ab4aaf863aea6a7d1e67e0c2ad536ec1f0c8ccf7/nav2_lifecycle_manager/src/lifecycle_manager.cpp#L264-L269, should say "failed to change node for planner server" after 
```cpp
!node_map_[node_name]->change_state(transition, std::chrono::milliseconds(-1),
      service_timeout_
```
receives the response after service server.  However, it seems that we are unable to get any response from the service server when running with rmw zenoh

JEnoch, 2025-09-10:
`rmw_zenoh` is by default configured with a [timeout of 60 seconds](https://github.com/ros2/rmw_zenoh/blob/bc6969ea3373f770ba0fc12de6613bd8aa9e8df2/rmw_zenoh_cpp/config/DEFAULT_RMW_ZENOH_SESSION_CONFIG.json5#L229) for queries (i.e. request to a Service).
According your log `z_reply_is_ok returned False Reason: Timeout`: this timeout is triggered, leading to just return from the callback and to not push any reply to `rcl` layer. See here: https://github.com/ros2/rmw_zenoh/blob/bc6969ea3373f770ba0fc12de6613bd8aa9e8df2/rmw_zenoh_cpp/src/detail/rmw_client_data.cpp#L413-420.

My understanding of RMW/RCL design is that it's not up to the RMW layer to send timeout error to the RCL layer. Actually, in `rmw.h` interface, the `rmw_take_response()` operation cannot return a timeout message, but only a valid and types `ros_response`: 
https://github.com/ros2/rmw/blob/db761272b97dff7ffced709dc2b33236af0aabbd/rmw/include/rmw/rmw.h#L2092
Therefore, the RCL layer manages an application timeout when `rclcpp::spin_until_future_complete()` is called with a timeout value: https://github.com/ros2/rclcpp/blob/364419589207059e3ef43e875134b0762d5222d2/rclcpp/include/rclcpp/client.hpp#L598

I believe that's exactly what Nav2 is doing in `ServiceClient::invoke()` here:
https://github.com/ros-navigation/navigation2/blob/8aea1a895f23ca505da3629819b6d16f4d126a0f/nav2_ros_common/include/nav2_ros_common/service_client.hpp#L109
However in your case the `timeout` is set to `std::chrono::milliseconds(-1)`, meaning infinite time. Thus I think the lifecycle manager remains blocked indefinitely because the service server didn't manage to send a reply before the 60 seconds `queries_default_timeout`.

Could you try to increase this config value to check if it make the test to pass ?

mini-1235, 2025-09-26:
> @mini-1235 https://github.com/ros2/rmw_zenoh/pull/820 should be a workaround for this issue, waiting for a definitive one addressing https://github.com/ros2/rmw/issues/404.

Thanks for the follow up @JEnoch!

> Why not using service_timeout_ for both transition_timeout and wait_for_service_timeout parameters in call to change_state() ?

I think this is a special case where we really need infinite timeout, as explained by @SteveMacenski in https://github.com/ros-navigation/navigation2/pull/4838#discussion_r2043325340

SteveMacenski, 2025-09-26:
Basically, due to other RMW's (at least historical, if not contemporary) quirks & arbitrary end-user code. Its the single only place in Nav2 where we do that.