ros2/rmw_zenoh #810: Build fails on ARM

mjforan, 2025-09-19:
### Operating System:

Linux 868aa2376466 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux

### ROS version or commit hash:

kilted

### Steps to reproduce issue

Building ROS 2 from source

```
ROS_DISTRO=kilted

apt install -y vcstool python3-rosdep2 python3-vcstools colcon build-essential

mkdir src && vcs import --input https://raw.githubusercontent.com/ros2/ros2/$ROS_DISTRO/ros2.repos src

rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO -y --skip-keys "fastcdr rti-connext-dds-7.3.0 urdfdom_headers python3-vcstool"

#  -DCMAKE_CXX_FLAGS="-Wno-maybe-uninitialized" due to this issue https://github.com/eProsima/Fast-DDS/issues/5277

 colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release  -DCMAKE_CXX_FLAGS="-Wno-maybe-uninitialized" --packages-up-to geometry2 launch launch_ros rcl rmw_cyclonedds_cpp rmw_implementation ros2cli_common_extensions image_common vision_msgs
```

### Expected behavior

Successful build

### Actual behavior

```
...lots of downloads...
  Downloaded tokio v1.40.0
  Downloaded sha3 v0.10.8
  Downloaded ring v0.17.13
  Downloaded sha2-const-stable v0.1.0
error: package `rustls-platform-verifier v0.3.4` cannot be built because it requires rustc 1.64.0 or newer, while the currently active rustc version is 1.63.0
Either upgrade to rustc 1.64.0 or newer, or use
cargo update -p rustls-platform-verifier@0.3.4 --precise ver
where `ver` is the latest version of `rustls-platform-verifier` supporting rustc 1.63.0
gmake[5]: *** [CMakeFiles/cargo.dir/build.make:146: release/target/release/libzenohc.so] Error 101
gmake[4]: *** [CMakeFiles/Makefile2:189: CMakeFiles/cargo.dir/all] Error 2
gmake[3]: *** [Makefile:166: all] Error 2
gmake[2]: *** [CMakeFiles/zenoh_c_vendor.dir/build.make:85: zenoh_c_vendor-prefix/src/zenoh_c_vendor-stamp/zenoh_c_vendor-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/zenoh_c_vendor.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zenoh_cpp_vendor [10min 51s, exited with code 2]
```

YuanYuYuan, 2025-09-19:
Hi @mjforan! Building rmw_zenoh from source requires a Rust toolchain newer than 1.75.0 installed.

mjforan, 2025-09-19:
Sorry, I should have looked more closely before posting.. 

Even on Ubuntu 24.04 the Apt version of cargo is 1.75.0. so on Debian I will have to install manually instead of through rosdep.

closed 2025-09-19 by mjforan